letscontrolit/ESPEasy #1340: MQTT#Connect

giig1967g, 2018-05-01:
Hi,
I would like to suggest the following improvement:
on MQTT#Connected the unit should send to MQTT broker the status of the subscribed topics.
Now this action is manual.

TD-er, 2018-05-01:
Should we store the last message to any topic and send that when MQTT gets connected?

giig1967g, 2018-05-01:
I think it's a good idea.

TD-er, 2018-05-01:
That's something that can be fixed quite easily in the Controller.ino.

giig1967g, 2018-05-01:
I can't find the best part of the code where to do the change.
Probably the best way would be to call the SensorSendTask(taskID) function.
That calls the PLUGIN_READ and the SendData functions.
Just afraid to the time it could take to loop through all the 12 tasks.

TD-er, 2018-05-01:
No, MQTT things are controlled in Controller.ino. So all has to be done is create a function there to do the actual publish and replace all publish calls to use that function.
Only thing is the import MQTT plugin, which is something that really has to be changed, because it keeps being an exception.
So I suggest to keep this feature on pause until we take care of the MQTT issues and that's a lot:

- Switch plugin has some specific stuff which doesn't belong there.
- Only one MQTT controller possible at the moment.
- MQTT import plugin is really a kludge, which should use a controller but then to set some subscriptions and receive data.
- Have some buffering at controller level to keep the last message and send it when connection has been restored. (regardless of the controller)

giig1967g, 2018-05-01:
ok. agree to leave in standby

TD-er, 2018-05-01:
It is one of the main issues to do after the wifi stuff is over.

TD-er, 2018-05-01:
You could think about how long the last value has to be kept.
Is it useful to keep the last measurement for let's say an hour? a day? 5 minutes?

giig1967g, 2018-05-01:
there shouldn't be an expiry.
The idea behind is: whenever the MQTT connects, it needs to publish the current values to the broker.
As, the other way around, it is in MQTT Import: whenever it connects it asks for the values from the broker.

TD-er, 2018-05-01:
So it is not a 'retained' value, but a loop over all plugins to send their measurements as  soon as the connection is restored?

giig1967g, 2018-05-03:
yes correct.

TD-er, 2019-10-27:
I've just read this issue again and I think it is just a matter of adding a command to trigger a PLUGIN_READ for all tasks.
At this moment we have `TaskRun`, so we only need to have a variant which either runs all or a list of task numbers.
Or maybe a command which does run all tasks assigned to a specific controller?

I think it is a bit more useful compared to sending the current state.
Problem is that some plugins may have incomplete states.
Also it is more useful to have all sensors triggered on a specific event. (e.g. location based interval events)

TD-er, 2022-04-01:
Should be fixed by these PRs: 
- https://github.com/letscontrolit/ESPEasy/pull/3987
- https://github.com/letscontrolit/ESPEasy/pull/3983

Please let me know if it is indeed fixed now... finally

closed 2022-04-01 by TD-er