letscontrolit/ESPEasy #1353: Domoticz HTTP Appears as Default First Controller - Disabled, but is Active

Oxyandy, 2018-05-02:
### Summarize of the problem/feature request
Domoticz HTTP Appears as Default First Controller - 
Disabled, but is Active when Domoticz MQTT is set to second controller.
Not sure of impact on other controllers. ?

### Expected behavior
On a new default Firmware all 3 controllers should be empty.
Controllers with 'Enable' unticked should not run code.

### Steps to reproduce
Flash ESP_Easy_mega-20180502 on a blank node.
Visit controllers
[image]

Oxyandy, 2018-05-02:
The quick fix, look into why it can not be disabled & remove as default, then close
The code maybe a mess, there will be a different issue that covers this.

TD-er, 2018-05-02:
It is now no longer the default when reset to factory default.
However, the second part of this issue, still remains, that its presence affects operations, even when disabled.

tonhuisman, 2022-04-18:
This controller seems to have been re-written, fixing this problem, so issue can be closed.

closed 2022-04-18 by TD-er