letscontrolit/ESPEasy #1355: Hostname is getting lost after a reboot

stoffelll, 2018-05-02:
### Summarize of the problem/feature request

If I set the hostname via serial console (in the Web-UI is no option to set the hostname), the hostname is being resetted after a reboot.

### Expected behavior

Keep hostname after a reboot

### Actual behavior

Hostname is resetted after  a reboot to its default value ("ESP-Easy-0")

### Steps to reproduce

1.  set hostname via console: Name <new name>
2.  reboot device

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

ghtester, 2018-05-02:
You need to enter Save command before reboot.

stoffelll, 2018-05-02:
Thanks for the hint with "save" and sorry for the not needed ticket!
I didn't saw the command in the list :/

closed 2018-05-02 by stoffelll