letscontrolit/ESPEasy #1356: esp8266 slows significantly when MQTT server unavailable/disconnected

ZericE, 2018-05-02:
### Summarize of the problem/feature request
When a working connection to an MQTT broker becomes unavailable OR a connection to a non-existent broker is attempted, the esp8266 slows down significantly from that point forward. If the MQTT server becomes available, the esp8266 goes back to normal. See below for more on the source of the problem and a possible code change.

### Expected behavior
The sudden unavailability of a connection to MQTT broker should not significantly disrupt all other operations of the esp device for an ongoing basis.

### Actual behavior
Getting a response from the web interface or through local gpio inputs becomes slow and unreliable when an unavailable MQTT broker is defined. After a few minutes, the "Load" spikes to 100% and stays there.

### Steps to reproduce
Create a MQTT controller in espeasy (I selected OpenHab controller),  point it to a non existing MQTT broker, enable it, and reboot. Usually it's possible to get the web interface to respond enough to disable the controller, although it may take more than one try.

### System configuration
Hardware: sonoff (basic and TH1x models). This isn't a hardware dependant issue.

ESP Easy version:  V2.0 Mar 22 2018 build and Mega May 02 2018 build. This has been an issue for quite some time. 

ESP Easy settings/screenshots: 
[image]
[image]
[image]

### Rules or log data
[image]

### Source of the problem
When pubsubclient attempts to connect to an unavailable MQTT broker via MQTTclient.connect(...) in Controller.ino, it blocks other esp operations for >6 seconds in V2.0 and >5s for Mega. Additionally, when a controller is unavailable, espeasy aggressively tries to (re)connect.

In ESPEasy.ino, the delay between MQTT connection attempts is set by timermqtt_interval and maxes at 2000 milliseconds delay (2s). This means that most of the CPU time (>70%) is spend trying to (re)connect to the broker, leaving little time for other activities. This long delay when an MQTT broker connect attempt fails is a known issue and occurs below the pubsubclient library from my understanding after reading the issue list for pubsubclient.

ESP Easy can however, resort to a far less aggressive (re)connect policy, leaving more time for other activities. There is little reason I can think of to attempt a reconnect more than a 1-3 times per minute after a couple of quicker tries. I have recompiled V2.0 changing the max timermqtt_interval from 2000ms to 30000ms and the backoff increment internal (currently: timermqtt_interval += 250) to a higher value of 5000 (5s), and that helps improve the situation significantly.  I no longer see the Load maxed out when it's trying to reconnect to an unavailable MQTT broker and in general the device is more responsive.

```

```

TD-er, 2018-05-02:
I passed the code you described, so I applied the suggested parameters.
Could you please verify them?

ZericE, 2018-05-03:
Great, your code update is  as I described. I'll try it on the next nightly build.

Oxyandy, 2018-05-03:
👍 for this, has been discussed before, looking forward to testing this on a 'real world' task

s0170071, 2018-05-03:
There is also a timeout setting for the WiFi client. For modbus I have set it from 5 s to 500ms. Client.connect is blocking, especially if the server is not available. @td-er what is the current timeout?

ZericE, 2018-05-03:
The change did not behave exactly as desired. I expected after a couple minutes when things settled down to see a regular retry rate of about every 30s, however that's not what happened. After it settled down, the time between retries seemed to go back and forth between ~10s and ~20s. Which led me to think that a MQTT broker reconnect attempt is happening somewhere else unexpectedly.

After some searching, I found that the following lines in function runEach30Seconds:
  int enabledMqttController = firstEnabledMQTTController();
  if (enabledMqttController >= 0) {
    MQTTCheck(enabledMqttController);
  } 

Checking MQTT here does not appear needed as it's redundant with runPeriodicalMQTT so I commented them all out and recompiled (I'm still using Core 2_3_0 core, but it shouldn't matter for this). This produced the expected behavior with a MQTT broker check ratcheting up by 5s until it reaches ~30s when it's disconnected.  When it's connected to the broker, I didn't see any issues so far.

I also noticed some lines in runPeriodicMQTT that may not be needed, so I may look into that further.

TD-er, 2018-05-03:
OK, keep me posted and please also try to test what happens when you boot the ESP while the MQTT broker is offline and gets online in a few minutes. (normal behavior after power-outage ;) )

ZericE, 2018-05-03:
Good idea @TD-er .  I tried two specific tests with the version I compiled (the one with lines removed from runEach30Seconds I mentioned in last msg).

First, I unplugged the broker and rebooted the client per your suggestion. The client started looking for the broker upon boot. I then plugged the broker back into the LAN a couple minutes later and the client connected as expected.

Next, with the client connected and working with the broker, I unplugged the broker from the LAN for a couple minutes. The client detected this and started it's reconnect process, ratcheting it up to 30s retries.  When I reconnected the broker, the client connected and resumed.

The responsiveness is definitely improved with this change when the broker is missing, and the load doesn't go to high levels anymore.

Oxyandy, 2018-05-03:
@ZericE - I really appreciate your effort. 
Example: I try to control a pump with a float switch, once fluid rises, reaches the float switch pump turns off.. Day after day it works fine..
Some MQTT connectivity issue, pump stays on too long, fluid rises above switch and bad things happen.
Following with great interest ;)

ZericE, 2018-05-04:
@Oxyandy - This change should help situations like yours where one needs to detect a relatively slow change in state.  For your pump, it should be fine as once the water is high enough to activate the float switch, it will stay activated and worst case the esp will see it just a few seconds later.

If a short duration change (pulse) happens on a gpio, such as a pushbutton, while the esp is busy trying to reconnect the mqtt broker, I expect it will be missed and a second pulse/button press would be needed.

Oxyandy, 2018-05-04:
@ZericE, my switch starts out '0', (Closed circuit)
then as it rises with the water level it changes state to '1', 
then very quickly the water level lowers & it returns to '0' again.
Many times could be considered just a pulse too.
Anyway for sure I will be investigating the impact these changes make to a dropped connection to MQTT

TD-er, 2018-05-04:
@Oxyandy A bit off-topic, but I was thinking about your use case.
You could also add a flip-flop between the ESP and the sensor and then connect the flipflop output and the sensor value to the ESP.
The flipflop would store the event of the change a bit longer. If the ESP detects a change in the flipflop state, then have a look at the true state of the sensor.
That would make detecting events a bit less time critical.
Still the bug you experience should be removed to deal with proper state handling.

Oxyandy, 2018-05-04:
Switch Logs show 2 seconds all up in the change on average..
I have a learnt heaps about the various causes as you have read, the biggest issue was the wrong state issue, I now know how to avoid that, still haven't updated firmware on 'real world' devices..
On never ending 'to do list'.........
I think my best solution is a second Sonoff basic with a float switch mounted higher as a 'watch dog' with a NC relay (opposite of original one) and it powers 'Main Sonoff', should fluid level reach second float switch, it stops power to main Sonoff, this will stop the pump & cold boot Main.. = never too much water
When I started I know nothing of IoT, I have evolved... slightly ;)

ZericE, 2018-05-07:
Tested the changes with the latest nightly and it's working much better under the affected conditions.  Would be nice to reduce the TCP connect timeout from 5 to something less like 2 for MQTT, but it's not a big issue.

closed 2018-05-07 by ZericE