letscontrolit/ESPEasy #1360: Do we have a memory leak in some plugins?

micropet, 2018-05-03:
The charts were made with the official test version of 29.04.2018.

The device (ESP-201) with only one Pir, a few Leds and a Neopicel Stripe runs for a few days without reboot.

The device (ESP-206), which still has a CO2 sensor MH-Z19 and a dust sensor PMS7003, reboots every few hours.
You can see that the Sysheap falls below 11000, then the reboot takes place.

[image]

[image]

micropet, 2018-05-03:
No, the power adapter makes 5V at 4A power.

On the device are 2x 1800μF capacitors, directly on the Wemos D1.
The voltage I have measured via a DSO, it is at 5.15 volts.

micropet, 2018-05-03:
Here you can see it even better

[image]

TD-er, 2018-05-03:
The BME280 is also in operations here at my sensors, so that's also confirmed to be working just fine.
Leaves the Neopixel, BH170 and the PMS sensor.
In the last screenshots it is indeed quite clear.

micropet, 2018-05-03:
Neopixel is indeed connected to the ESP-201.
That does not cause any problems.

micropet, 2018-05-03:
Here ist only an SD021 connected.

[image]

micropet, 2018-05-05:
Something has gotten better. 
This is the device with Pir, Led, BME280, BH170, Neopixel, CO2 MH-Z19, Dust PMS7003

[image]

closed 2022-04-21 by TD-er