letscontrolit/ESPEasy #1362: Secondary DNS set to 0.0.0.0

giig1967g, 2018-05-03:
Hi,
I just noticed that in the info page the DNS settings are reported as follows.
Does it mean that the secondary DNS is 0.0.0.0?
Can it generate errors? 

Network ❔
--
Wifi | 802.11N (RSSI -66 dB)
IP config | Static
IP / subnet | 192.168.1.206 / 255.255.255.0
GW | 192.168.1.1
Client IP | 192.168.1.79
DNS | 8.8.8.8 / 0.0.0.0
Allowed IP Range | 192.168.1.0 - 192.168.1.255

TD-er, 2018-05-03:
Currently it cannot be set via the config.
But I was looking into retrieving data from the DHCP to be able to set it.
As far as I know, the DNS requests are not sent to an 'invalid' IP-address as 0.0.0.0

giig1967g, 2018-05-03:
can we set it to 8.8.4.4 or 1.0.0.1 ?

TD-er, 2018-05-03:
Setting to 1.0.0.1 **will** lead to problems, since that one is a valid IP address.
And if someone only wants to use an internal DNS, to be able to resolve internal hosts, setting a hard coded external DNS which cannot be changed is also a receipt for new issues.

The current solution will only lead to issues when the first one is unreachable.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er