letscontrolit/ESPEasy #1363: [UI] Checkbox overlap the header when scrolling page.

workgroupengineering, 2018-05-03:
In advanced settings, when scrolling page the checkbox overlap the header.

### Expected behavior

 Checkbox do not overlap the header.

### Actual behavior
 Checkbox  overlap the header.
[image]

### Steps to reproduce

1.  on 19" screen with resolution 1224x768 
2.  go  advanced settings http://http://192.168.1.x/advanced
3.  scrolling page

### System configuration
|                  |                                  |
| ------------- | ------------- |
| Hardware | **Sonoff  TH10/16** |
| ESP Easy version | **20102 - Mega** |
| Libraries | **ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 1.4.0-RC2** |
| GIT version | **mega-20180503** |
| Plugins  | **72 [Normal] [Testing] [Development]** |
| Build Md5 | **605586a9cf8191c86ba991dda2c61ff** |
| Build time | **May  3 2018 02:07:06** |
| Binary filename | **firmware.bin** |

Grovkillen, 2018-05-03:
Aaaah! Need to fix that! 😉

Grovkillen, 2018-05-03:
Fixed

closed 2018-05-03 by Grovkillen