letscontrolit/ESPEasy #1368: Crash Plugin Output - NeoPixel (Word Clock)

cauer71, 2018-05-04:
If i enable the Plugin Output - NeoPixel (Word Clock), espeasy crashes every time, this issue is reproducible

TD-er, 2018-05-04:
Is the neopixel connected?
And is it working in the other modes (other than world clock)?

Do you have a log output?

cauer71, 2018-05-04:
the neopixel is connected, and in the basic plugin works fine. u can simply try to add the word clock Task and enable it. i dont get log, because the esp reboots immediately

ZericE, 2018-05-05:
@cauer71 I tried it with the serial monitor running and it hard crashes with Exception (28), which I believe means access to an invalid address was attempted. Nothing appeared of interest in the log prior.

ZericE, 2018-05-05:
The line that is causing the crash is:  **Plugin_041_pixels->setPixelColor(i, Plugin_041_pixels->Color(0, 0, 0));**
in the function **resetAndBlack**, called from **Plugin_041_update**, called from **PLUGIN_WEBFORM_SAVE**.

The library Adafruit_NeoPixel hasn't been initialized yet in the plugin when the Webform is saved so it's not surprising it abends. Calling **Plugin_041_update** from **PLUGIN_WEBFORM_SAVE** isn't strictly necessary, it's more aesthetic to clear the display, and removing it fixes the crash. The code to initialize the library when the webform is saved could be added if it's important to clear the display at that time(?). Unfortunately I don't have any way to test the overall functionality as I don't have the hardware.

@TD-er what is the protocol for this situation so a built version can be given to @cauer71 to test with his hardware?

TD-er, 2018-05-05:
If it is just to test, build one yourself and provide a link I guess :)
And maybe you could provide a pull request?

ZericE, 2018-05-05:
@TD-er I figured it would be better to do a pull request after someone has tested it with actual hardware. I did put in a few temporary debugging log entries in my testing and it appears to be cycling through the functions the way I would expect, but rather it displays correctly is another story (and not one I'll likely be able to fix without the HW).

@cauer71 I can put a build up someplace for you.  What memory size do you need, 1M?  I'm assuming you are using the esp8266 NORMAL version.

cauer71, 2018-05-06:
@ZericE i use a wemos mini d1 with 4M

ZericE, 2018-05-06:
Give this a try and let me know.  I did a quick test on a 4M Mini D1 and it doesn't crash for me.

[ESP Easy Word Clock Test ](https://drive.google.com/open?id=1wff9NtRLYM1IXCBoMZB9PfQxbWJOlQBn)

Oxyandy, 2018-05-06:
While looking over compiler warnings - I noticed that the Lib for this has had some updates..
Data latch is slowed, extra lines for ESP32 + others - anyhow, not sure if ESPeasy included Lib needs an update. ?
https://github.com/adafruit/Adafruit_NeoPixel

ZericE, 2018-05-07:
@Oxyandy Makes sense on updating the library.  All I was able to do for this issue was fix the crash, the plugin may or may not work properly since I don't have the hardware to test it.

I'm not sure why this plugin is part of the Normal build, maybe I'm wrong, but it seems like it has a rather limited audience.

As far as I can tell, the code has been broken for a long time. I had a binary from Jan 2, 2018 laying around that I tried and it also has this crash bug, but no one has reported it in the last four months until now.

cauer71, 2018-05-07:
It does not crash anymore, can you provide me the updatetd _P041_NeoClock.ino so i can adapt it to the german layout of the clock?

TD-er, 2018-05-07:
And create a PR for the fix :)

Oxyandy, 2018-05-07:
ZericE, nice work

ZericE, 2018-05-07:
@cauer71 in _P041_NeoClock.ino, within case PLUGIN_WEBFORM_SAVE, just delete the line that reads: Plugin_041_update(). It's around line number 69. That's it.
[...]
case PLUGIN_WEBFORM_SAVE:
      {
        Settings.TaskDevicePluginConfig[event->TaskIndex][0] = getFormItemInt(F("plugin_041_red"));
        Settings.TaskDevicePluginConfig[event->TaskIndex][1] = getFormItemInt(F("plugin_041_green"));
        Settings.TaskDevicePluginConfig[event->TaskIndex][2] = getFormItemInt(F("plugin_041_blue"));
        Plugin_041_red = Settings.TaskDevicePluginConfig[event->TaskIndex][0];
        Plugin_041_green = Settings.TaskDevicePluginConfig[event->TaskIndex][1];
        Plugin_041_blue = Settings.TaskDevicePluginConfig[event->TaskIndex][2];
        **Plugin_041_update();  <--- DELETE THIS ENTIRE LINE**
        success = true;
[...]

@TD-er I will create a PR.

TD-er, 2021-10-14:
Already fixeed

closed 2021-10-14 by TD-er