letscontrolit/ESPEasy #1372: Command timerSet,*,0 does not reset timer

iBobik, 2018-05-05:
### Summarize of the problem/feature request

1. timerSet,1,5
2. timerSet,1,0
Timer was triggered.

### System configuration

Hardware: ESP-01 1024

ESP Easy version: ESPEasy_R120_1024.bin

ESP Easy settings/screenshots: 

### Rules or log data
Used Rules code from the example: https://www.letscontrolit.com/wiki/index.php/Tutorial_Rules#Starting.2Fstopping_repeating_timers_with_events

### Workaround
Use `timerSet,2,999999999` - it will not stop timer forever, but set it to ~31 years (biggest possible 9s value).

ZericE, 2018-05-05:
R120 is quite old and doesn't get much attention from developers. Have you tried a newer build, like v2.0 from Mar 22, 2018 to see if this is still an issue?

ZericE, 2018-05-06:
@iBobik I tested for this timerSet issue in the v2.0 and on the Mega versions, both correctly stop the timer when it's set to 0.

There hasn't been any changes to R1XX branch since February 2017 and it's not likely to get any bug fixes. V2.0 has been more stable for me than R120 was, you may want to migrate to it.

iBobik, 2018-05-06:
Upgraded to current mega version and it seems working.

closed 2018-05-06 by iBobik