letscontrolit/ESPEasy #1375: Compilers Warnings

Oxyandy, 2018-05-06:
What to do about these, if warnings exist on my local builds, how do they impact GitHub builds ?
Adafruit_TSL2591.cpp
```
Compiling .pioenvs\normal_ESP8266_1024\libf19\Adafruit_TSL2591\Adafruit_TSL2591.cpp.o
lib\Adafruit_TSL2591\Adafruit_TSL2591.cpp: 
In member function 'float Adafruit_TSL2591::calculateLuxf(uint16_t, uint16_t)':
lib\Adafruit_TSL2591\Adafruit_TSL2591.cpp:178:12: 
warning: unused variable 'chan0' [-Wunused-variable]
uint32_t chan0, chan1;
^
lib\Adafruit_TSL2591\Adafruit_TSL2591.cpp:178:19: 
warning: unused variable 'chan1' [-Wunused-variable]
uint32_t chan0, chan1;
^
lib\Adafruit_TSL2591\Adafruit_TSL2591.cpp: 
In member function 'uint32_t Adafruit_TSL2591::getFullLuminosity()':
lib\Adafruit_TSL2591\Adafruit_TSL2591.cpp:274:64: 
warning: 'y' may be used uninitialized in this function [-Wmaybe-uninitialized]
y |= read16(TSL2591_COMMAND_BIT | TSL2591_REGISTER_CHAN0_LOW);
^
```
Adafruit_NeoPixel.cpp
```
Compiling .pioenvs\normal_ESP8266_1024\lib8da\Adafruit_NeoPixel\Adafruit_NeoPixel.cpp.o
lib\IRremoteESP8266\IRremoteESP8266.cpp: 
In member function 'void IRrecv::copyIrParams(irparams_t*)':
lib\IRremoteESP8266\IRremoteESP8266.cpp:818:36: 
warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
for (int i=0; i<sizeof(irparams_t); i++)
^
```
Adafruit_MS_PWMServoDriver.cpp
```
Compiling .pioenvs\normal_ESP8266_1024\lib548\Adafruit_Motor_Shield_V2\utility\Adafruit_MS_PWMServoDriver.cpp.o
lib\Adafruit_Motor_Shield_V2\Adafruit_MotorShield.cpp: 
In member function 'void Adafruit_StepperMotor::step(uint16_t, uint8_t, 
uint8_t)':
lib\Adafruit_Motor_Shield_V2\Adafruit_MotorShield.cpp:220:11: 
warning: variable 'ret' set but not used [-Wunused-but-set-variable]
uint8_t ret = 0;
^
lib\Adafruit_Motor_Shield_V2\Adafruit_MotorShield.cpp: 
In member function 'uint8_t Adafruit_StepperMotor::onestep(uint8_t, uint8_t)':
lib\Adafruit_Motor_Shield_V2\Adafruit_MotorShield.cpp:242:11: 
warning: unused variable 'a' [-Wunused-variable]
uint8_t a, b, c, d;
^
lib\Adafruit_Motor_Shield_V2\Adafruit_MotorShield.cpp:242:14: 
warning: unused variable 'b' [-Wunused-variable]
uint8_t a, b, c, d;
^
lib\Adafruit_Motor_Shield_V2\Adafruit_MotorShield.cpp:242:17: 
warning: unused variable 'c' [-Wunused-variable]
uint8_t a, b, c, d;
^
lib\Adafruit_Motor_Shield_V2\Adafruit_MotorShield.cpp:242:20: 
warning: unused variable 'd' [-Wunused-variable]
uint8_t a, b, c, d;
^
lib\Adafruit_Motor_Shield_V2\Adafruit_MotorShield.cpp: 
In member function 'Adafruit_DCMotor* Adafruit_MotorShield::getMotor(uint8_t)':
lib\Adafruit_Motor_Shield_V2\Adafruit_MotorShield.cpp:89:31: 
warning: 'in2' may be used uninitialized in this function [-Wmaybe-uninitialized]
dcmotors[num].IN2pin = in2;
^
lib\Adafruit_Motor_Shield_V2\Adafruit_MotorShield.cpp:88:31: 
warning: 'in1' may be used uninitialized in this function [-Wmaybe-uninitialized]
dcmotors[num].IN1pin = in1;
^
lib\Adafruit_Motor_Shield_V2\Adafruit_MotorShield.cpp:87:31: 
warning: 'pwm' may be used uninitialized in this function [-Wmaybe-uninitialized]
dcmotors[num].PWMpin = pwm;
^
lib\Adafruit_Motor_Shield_V2\Adafruit_MotorShield.cpp: 
In member function 'Adafruit_StepperMotor* Adafruit_MotorShield::getStep
per(uint16_t, uint8_t)':
lib\Adafruit_Motor_Shield_V2\Adafruit_MotorShield.cpp:118:33: 
warning: 'bin2' may be used uninitialized in this function [-Wmaybe-uninitialized]
steppers[num].BIN2pin = bin2;
^
lib\Adafruit_Motor_Shield_V2\Adafruit_MotorShield.cpp:117:33: 
warning: 'bin1' may be used uninitialized in this function [-Wmaybe-uninitialized]
steppers[num].BIN1pin = bin1;
^
lib\Adafruit_Motor_Shield_V2\Adafruit_MotorShield.cpp:116:33: 
warning: 'ain2' may be used uninitialized in this function [-Wmaybe-uninitialized]
steppers[num].AIN2pin = ain2;
^
lib\Adafruit_Motor_Shield_V2\Adafruit_MotorShield.cpp:115:33: 
warning: 'ain1' may be used uninitialized in this function [-Wmaybe-uninitialized]
steppers[num].AIN1pin = ain1;
^
lib\Adafruit_Motor_Shield_V2\Adafruit_MotorShield.cpp:114:33: 
warning: 'pwmb' may be used uninitialized in this function [-Wmaybe-uninitialized]
steppers[num].PWMBpin = pwmb;
^
lib\Adafruit_Motor_Shield_V2\Adafruit_MotorShield.cpp:113:33: 
warning: 'pwma' may be used uninitialized in this function [-Wmaybe-uninitialized]
steppers[num].PWMApin = pwma;
^
```
AS_BH1750A.cpp
```
Compiling .pioenvs\normal_ESP8266_1024\liba15\AS_BH1750\AS_BH1750A.cpp.o
lib\AS_BH1750\AS_BH1750A.cpp: 
In member function 'bool AS_BH1750A::startMeasurementAsync(TimeFuncPtr)':
lib\AS_BH1750\AS_BH1750A.cpp:518:7: 
warning: unused variable '_lastResult' [-Wunused-variable]
int _lastResult = -1;
^
lib\AS_BH1750\AS_BH1750A.cpp: In member function 'bool AS_BH1750A::delayExpired()':
lib\AS_BH1750\AS_BH1750A.cpp:533:19: 
warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
if (timestamp < _lastTimestamp)
^
lib\AS_BH1750\AS_BH1750A.cpp:536:45: 
warning: statement has no effect [-Wunused-value]
delayTime = MAX_U_LONG - _lastTimestamp + timestamp;
^
lib\AS_BH1750\AS_BH1750A.cpp:542:24: 
warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
return (delayTime >= _nextDelay);
^
```

TD-er, 2018-05-06:
These should only affect plugins using these libraries and most of these warnings are quite harmless.
A few can be rather serious, like uninitialised variable.
But they should not affect WiFi stability.

Oxyandy, 2018-05-06:
Yeah, I know that this has nothing to do with Wifi.. I was thinking GitHub build were done by a bot & wondered if 'warnings' effected it in anyway.. but now find out it's a human..
Now can say with all confidence there is a difference between the GitHub packaged & zipped 'Releases' compared to Self-Compiled, but why... ?

Oxyandy, 2018-05-06:
Are these the homes of these libs ?
Some have changed I guess, IRremoteESP8266 seems to have more devices added etc
[Adafruit_TSL2591_Library](https://github.com/adafruit/Adafruit_TSL2591_Library)
[Adafruit_Motor_Shield_V2_Library](https://github.com/adafruit/Adafruit_Motor_Shield_V2_Library)
[IRremoteESP8266](https://github.com/markszabo/IRremoteESP8266) 
[AS_BH1750](https://github.com/hexenmeister/AS_BH1750)

Oxyandy, 2018-05-06:
Ok so I will leave this Open as a reminder ?

TD-er, 2018-05-06:
That's OK. :)

Oxyandy, 2018-05-07:
Would this style of accessing plugin source be any good?
Help with keeping Libs upto date... ?
https://platformio.org/lib/show/439/BH1750/installation

TD-er, 2018-05-07:
At this moment we have most used plugins included in our source tree.
This has a number of reasons:

1. It is guaranteed to be stable.
2. Arduino IDE users do not 'look' at the fixed versions in PlatformIO.ini.
3. Sometimes the versions we use have local changes to make them work with ESPeasy or patch some bugs which are not (yet) fixed in the main repository of that library.

For example the OLED library has only very recently been updated to 4.0, and now includes all patches I made to fix memory leaks, uninitialized variables and some other fixes. Along with those fixes, the 4.0 version has some nice new updates I would like to include in ESPeasy (like 255 steps dimming and multi display support).
But I will test it first on my own nodes, since the OLED plugin is one of the most used plugins and there have been many issues with memory leaks in the past. So I will go through that library to check all new code.

The IR library has been updated a while ago in the main repository, but that update was also a major upgrade in supported protocols and thus in code size.

In short: Yes we would like to upgrade, but in a controlled way. Don't fix what isn't broken :)

tonhuisman, 2023-09-03:
This seems to be (mostly) solved, so can be closed.

closed 2023-09-04 by TD-er