letscontrolit/ESPEasy #1376: Feature Request SERVO for ESP32

waspie, 2018-05-06:
Best I can tell the standard arduino servo library will not work with the ESP32.
I have a need to use 2 servos that I used to use with the esp8266

https://www.arduinolibraries.info/libraries/esp32-servo

# Servo Library for ESP32

This library attempts to faithfully replicate the semantics of the
Arduino Servo library (see http://www.arduino.cc/en/Reference/Servo)
for the ESP32, with two (optional) additions. The two new functions
expose the ability of the ESP32 PWM timers to vary timer width.

waspie, 2018-05-15:
I fumbled my way through making some changes to the switch plugin like including the esp32 servo library and changing the ifdefs from ESP8266 to ESP32 and compiled a bin and it does work
From my perspective there aren't any major changes. Just include the esp32 servo lib and make it so servo usage isnt exclusive to the esp8266. Sorry, I'm not talented enough to know how to do all this, just enough to "hack" (in the sense of patching it together with duct tape) what's already there to work for me.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er