letscontrolit/ESPEasy #1379: Check task/value names for "unwanted" characters + adding brackets in rules

Grovkillen, 2018-05-06:
Today we have little to no lookup on names of tasks and those tasks value names. This makes a rule like this a bit hard to parse for the rules engine:

```
on DHT11 Outside#Temperature>20 do
GPIO,2,1
endon
```
For formulas in rules we enclose these with square brackets, my opinion is that we should always use this for events as well. The above example would then look like this:

```
on [DHT11 Outside#Temperature]>20 do
GPIO,2,1
endon
```

That way we could allow for spaces in names (which I think is cool because you might want to inject those value names into a homepage for example). But we should also make sure that we never have a hashtag inside a name. What else should not be allowed?

Open discussion

TD-er, 2018-05-06:
Quotes should also be left out.
I guess also multi-byte UTF-8 characters may be causing issues when parsing.

giig1967g, 2018-05-07:
What about backwards compatibility?
Wouldn't this change break all setups?

Grovkillen, 2018-05-07:
Yes it probably will but should instead trigger a warning. We need to take care of this ahead of time I think.

workgroupengineering, 2018-05-07:
In a private or I'm working. I used a similar syntax but only on the name.
```
on [DHT11 Outside]#Temperature>20 do
GPIO,2,1
endon
```
in this way there are no backward compatibility problems.

closed 2022-04-21 by TD-er