letscontrolit/ESPEasy #1388: 60 second timerSet does not work.

vandalon, 2018-05-08:
### Summarize of the problem/feature request

When using SetTimer for 60 seconds it just won't run, using 5 or 59 works just fine.
### Expected behavior

Timer to run after 60 seconds ;)

### Actual behavior
Timer rule does not get executed.

### Steps to reproduce
1. Create a timer rule with some functions
2. set the timer for 60 seconds see it not working.
3. change the timer to 5 or 59 seconds, see it working :)

Problem stays after poweroff.

### System configuration

Hardware: Wemos d1 mini

ESP Easy version: ESP_Easy_mega-20180507

ESP Easy settings/screenshots: 

### Rules or log data

```On afzuiging#afzuiging=0 do
   timerSet,1,3600
   timerSet,2,60
endon

On Rules#Timer=2 do
   TaskValueSet 7,2,[dht#humidity]
   TaskValueSet 8,4,[humHistory#three]
   TaskValueSet 8,3,[humHistory#two]
   TaskValueSet 8,2,[humHistory#one]
   TaskValueSet 8,1,[dht#humidity]
   timerSet,2,60
endon
```

ZericE, 2018-05-08:
The following rules use timers 1 and 2 at 60 seconds to cycle an on-board LED.  It works correctly, so there is something else coming into play (ver mega-20180507).  I'll play around with it using a Dummy Device.

On System#Boot do
  gpio,2,0
  timerSet,2,60
endon

On Rules#Timer=1 do
  gpio,2,0
  timerSet,2,60
endon
On Rules#Timer=2 do
  gpio,2,1
  timerSet,1,60
endon

ZericE, 2018-05-09:
@vandalon - The following rules, which are similar to yours using dummy devices in task 7 and 8, do not produce your error.  I can watch the values cycling up in the log every 60s.  Can you add some screen shots showing a list of all your devices, and details on the devices referenced in your rules? Any other thoughts?

```
On System#Boot do
   timerSet,1,3600
   timerSet,2,60
endon

On Rules#Timer=2 do
   TaskValueSet 7,2,[ddev7#dv2]+1
   TaskValueSet 8,4,[ddev8#dv4]+1
   TaskValueSet 8,3,[ddev8#dv3]+1
   TaskValueSet 8,2,[ddev8#dv2]+1
   TaskValueSet 8,1,[ddev8#dv1]+1
   timerSet,2,60
endon
```

TD-er, 2018-05-09:
It may be that a lot of plugins use 60 seconds interval, which takes more than 1 second to process and the rules timer does look whether something has to be done now, not if a timer has expired.

vandalon, 2018-05-09:
I am unable to reproduce the problem today, all timers with 60 seconds now work fine it seems.
here is a full rule set and a screenshot of the devices:

```
On System#Boot do
   timerSet,8,60
   if [afzuiging#afzuiging]=0
      gpio,16,0
   endif
endon

On MQTT#Connected do
   Publish home/%sysname%/motion,0
endon

On afzuiging#afzuiging=0 do
   TaskValueSet 7,3,[humidity#prevHum]
   timerSet,1,3599
   timerSet,2,60
endon

On afzuiging#afzuiging=1 do
   TaskValueSet 7,3,0
   timerSet,1,0
   timerSet,2,0
endon

On Rules#Timer=1 do
   gpio,16,1
endon

On Rules#Timer=2 do
   timerSet,2,60
endon

On Rules#Timer=8 do
   TaskValueSet 7,1,[dht#humidity]
   TaskValueSet 8,4,[humHistory#three]
   TaskValueSet 8,3,[humHistory#two]
   TaskValueSet 8,2,[humHistory#one]
   TaskValueSet 8,1,[dht#humidity]
   TaskValueSet 7,2,([humHistory#one]+[humHistory#two]+[humHistory#three]+[humHistory#four]+[dht#humidity])/5
   timerSet,8,60
endon

On dht#humidity<90 do
   if [humidity#avgHum]<[humidity#startHum] and [humidity#startHum]<85
      gpio,16,1
      timerSet,1,0
      timerSet,2,0
   endif
   if [humidity#startHum]>84 and [humidity#avgHum] <90
      gpio,16,1
      timerSet,1,0
      timerSet,2,0
   endif
endon

On motion#sensor=1 do
   timerSet 3,0
   Publish home/%sysname%/motion,1
endon

On motion#sensor=0 do
   timerSet 3,30
endon

On Rules#Timer=3 do
   Publish home/%sysname%/motion,0
   timerSet 3,0
endon
```
[image]

ZericE, 2018-05-09:
It's going to be hard to find the problem if it's intermittent. Also, you may want to check you rules logic. It appears timer 2 isn't needed as nothing depends on it, and once gpio 16 is set to 1, I don't see anyway for it to be reset to 0.

What are you trying to do, run a fan in the W.C.?

vandalon, 2018-05-09:
It might be that indeed something took to long (more then 1 second). it would be nice if you could say something like ```if [timer2#value  > 60]```...
You are right about timer2. It became obsolete , just did not delete it. The logic in timer 8 I did in timer 2 before.
Gpio state 1 is actually state off, not on. it is set on remotely by home assistant.

uzi18, 2019-09-04:
@vandalon your rules are wrong, you cannot use "<" in section on...do...endon, but you can add another if just after :)

Grovkillen, 2019-10-10:
This isn't an issue.

closed 2019-10-10 by Grovkillen