letscontrolit/ESPEasy #1392: Copy as markdown firmware informations

workgroupengineering, 2018-05-09:
On firmware section of sysinffo add the possibilty to copy firmware informations (Build, Libraries, ecc..) in markdown.

### Expected behavior

Click on firmware header or button copy firmware informations on clipboard in markdown format.

### Actual behavior
none

Hardware: All
ESP Easy version: All

Grovkillen, 2018-05-09:
Might be fixed with #1393 ?

Grovkillen, 2018-05-10:
Could you please try the latest mega release? Report back here :+1:

workgroupengineering, 2018-05-10:
It's very good, but what I requested is different.
I would like to copy the firmware information in markdown to paste it without changes in a new issue or PR.

Grovkillen, 2018-05-10:
Maybe a GitHub copy button?

Grovkillen, 2018-05-15:
This is fixed as per #1432

closed 2018-05-17 by Grovkillen