letscontrolit/ESPEasy #1394: loadFromFile

s0170071, 2018-05-09:
When setting the log to debug after a connect it says
`LoadFromFile: config.dat index: 4096 datasize: 472 
`
all the time

Build
20102  - Mega
Libraries
ESP82xx Core 41a64707, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3
GIT version
(custom)
Plugins
49 [Normal]
Build time
May  9 2018 21:42:33
Binary filename
ThisIsTheDummyPlaceHolderForTheBinaryFilename...

INIT : Booting version: (custom) (ESP82xx Core 41a64707, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
88 : INIT : Warm boot #5
90 : FS   : Mounting...
115 : FS   : Mount successful, used 75802 bytes of 957314
127 : CRC  : No program memory checksum found. Check output of crc2.py
155 : CRC  : SecuritySettings CRC   ...OK 
259 : INIT : Free RAM:27136
259 : INIT : I2C
259 : INIT : SPI not enabled
352 : INFO : Plugins: 49 [Normal] (ESP82xx Core 41a64707, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
353 : WIFI : Set WiFi to STA
384 : WIFI : Connecting MNET attempt #0
407 : WD   : Uptime 0 ConnectFailures 0 FreeMem 23952
411 : Power: 0.00 Energy: inf
2399 : Power: 0.00 Energy: inf
3305 : WIFI : Connected! AP: MNET (5C:49:79:28:0C:6F) Ch: 1 Duration: 2920 ms
5095 : WIFI : DHCP IP: 192.168.1.44 (ESP-Easy-SMA-9) GW: 192.168.1.1 SN: 255.255.255.0   duration: 1789 ms
5108 : Webserver: start
5205 : MQTT : Intentional reconnect
5206 : LoadFromFile: config.dat index: 28672 datasize: 336
5246 : MQTT : Connected to broker with client ID: ESPClient_2C:3A:E8:06:5E:B2
5246 : Subscribed to: SMA
5399 : LoadFromFile: config.dat index: 4096 datasize: 472
5402 : LoadFromFile: config.dat index: 6144 datasize: 472
5404 : LoadFromFile: config.dat index: 4096 datasize: 472
5406 : LoadFromFile: config.dat index: 6144 datasize: 472
5410 : LoadFromFile: config.dat index: 4096 datasize: 472
5415 : LoadFromFile: config.dat index: 6144 datasize: 472
5428 : LoadFromFile: config.dat index: 4096 datasize: 472
5429 : LoadFromFile: config.dat index: 6144 datasize: 472
5432 : LoadFromFile: config.dat index: 4096 datasize: 472
5435 : LoadFromFile: config.dat index: 6144 datasize: 472
5441 : LoadFromFile: config.dat index: 4096 datasize: 472
5446 : LoadFromFile: config.dat index: 6144 datasize: 472
7398 : LoadFromFile: config.dat index: 4096 datasize: 472
7402 : LoadFromFile: config.dat index: 6144 datasize: 472
7404 : LoadFromFile: config.dat index: 4096 datasize: 472
7406 : LoadFromFile: config.dat index: 6144 datasize: 472
7410 : LoadFromFile: config.dat index: 4096 datasize: 472
7415 : LoadFromFile: config.dat index: 6144 datasize: 472
7428 : LoadFromFile: config.dat index: 4096 datasize: 472
7431 : LoadFromFile: config.dat index: 6144 datasize: 472
7433 : LoadFromFile: config.dat index: 4096 datasize: 472
7435 : LoadFromFile: config.dat index: 6144 datasize: 472
7441 : LoadFromFile: config.dat index: 4096 datasize: 472
7446 : LoadFromFile: config.dat index: 6144 datasize: 472
9399 : LoadFromFile: config.dat index: 4096 datasize: 472
9402 : LoadFromFile: config.dat index: 6144 datasize: 472
9405 : LoadFromFile: config.dat index: 4096 datasize: 472
9406 : LoadFromFile: config.dat index: 6144 datasize: 472
9411 : LoadFromFile: config.dat index: 4096 datasize: 472
9416 : LoadFromFile: config.dat index: 6144 datasize: 472
9428 : LoadFromFile: config.dat index: 4096 datasize: 472
9430 : LoadFromFile: config.dat index: 6144 datasize: 472
9432 : LoadFromFile: config.dat index: 4096 datasize: 472
9436 : LoadFromFile: config.dat index: 6144 datasize: 472
9442 : LoadFromFile: config.dat index: 4096 datasize: 472
9446 : LoadFromFile: config.dat index: 6144 datasize: 472
10397 : LoadFromFile: config.dat index: 4096 datasize: 472
10399 : Power: 0.00 Energy: inf
10477 : Invalid value detected for controller OpenHAB MQTT
10710 : Current Time Zone:  DST time start: 2018-03-25 02:00:00 offset: 60 minSTD time start: 2018-10-28 03:00:00 offset: 0 min
10731 : connect: 192.168.1.41 OK, sending read request: 0 1 0 0 0 6 3 4 119 53 0 4 
10816 : reading bytes: 0 1 0 0 0 11 3 4 8 > 0 0 0 0 0 0 133 92 value: 34140.00
11397 : LoadFromFile: config.dat index: 6144 datasize: 472
11401 : LoadFromFile: config.dat index: 4096 datasize: 472
11403 : LoadFromFile: config.dat index: 6144 datasize: 472
11405 : LoadFromFile: config.dat index: 4096 datasize: 472
11410 : LoadFromFile: config.dat index: 6144 datasize: 472
11415 : LoadFromFile: config.dat index: 4096 datasize: 472
11419 : LoadFromFile: config.dat index: 6144 datasize: 472
11432 : LoadFromFile: config.dat index: 4096 datasize: 472
11434 : LoadFromFile: config.dat index: 6144 datasize: 472
11436 : LoadFromFile: config.dat index: 4096 datasize: 472
11440 : LoadFromFile: config.dat index: 6144 datasize: 472
11446 : LoadFromFile: config.dat index: 4096 datasize: 472
11450 : LoadFromFile: config.dat index: 6144 datasize: 472

Grovkillen, 2018-05-09:
Yes , I noticed that too. Could be that we have a faster loop these days?

s0170071, 2018-05-09:
Why and what is it loading anyway?

uzi18, 2018-05-09:
Question is why it is reloaded every time ... and waste time.
If I remember correctly with config save also there was such issue, that something is written often, is it realy needed?

ZericE, 2018-05-09:
LoadFromFile in this case appears to be called from LoadControllerSettings, which is called by every controller.  Try disabling MQTT and any other controllers, see if changes the behavior.

TD-er, 2018-05-09:
I added those logs to get an idea of how often things are loaded and saved to flash.
It was with the intention to fix a bug that corrupts settings, which is already fixed by @mvdbro 

About loading from flash.
If I remember correctly, some plugins do read all extra configuration from flash every time they run.
Just to save RAM.
For some you may argue whether that's a good idea. But after all, all strings marked with `F()` are read from flash, so it isn't a bad idea :)

I was planning to look into memory usage anyway these days, so I will also have a look at these load from file things.

tonhuisman, 2022-05-07:
This doesn't seem to be relevant anymore, can be closed.

closed 2022-05-07 by TD-er