letscontrolit/ESPEasy #1396: Refactoring PlugIn ad Settings

workgroupengineering, 2018-05-10:
### Summarize of the problem/feature request

At this moment there are three / four different structures for managing plugins:

``` c
byte Plugin_id[PLUGIN_MAX];                     // is used to derive the pointer index for the Plugin_xxx function
boolean (*Plugin_ptr[PLUGIN_MAX])(byte, struct EventStruct*, String&);            // where store pointer to function Plugin_xxx
struct DeviceStruct
{
  DeviceStruct() :
    Number(0), Type(0), VType(0), Ports(0),
    PullUpOption(false), InverseLogicOption(false), FormulaOption(false),
    ValueCount(0), Custom(false), SendDataOption(false), GlobalSyncOption(false),
    TimerOption(false), TimerOptional(false), DecimalsOnly(false) {}
  byte Number;
  byte Type;
  byte VType;
  byte Ports;
  boolean PullUpOption;
  boolean InverseLogicOption;
  boolean FormulaOption;
  byte ValueCount;
  boolean Custom;
  boolean SendDataOption;
  boolean GlobalSyncOption;
  boolean TimerOption;
  boolean TimerOptional;
  boolean DecimalsOnly;
} Device[DEVICES_MAX + 1];   // contains plugin information

```
there are other fields in SettingsStruct that contain the link between the task and the plugin.

to get all the information on a plugin you need to iterate the devices array  until device.Number is not the same as what you are looking for, then you have to scan the Plugin_id array until the content is the same as the one searched and with the index just got it as an index of Plugin_ptr [] to get the function pointer.

in addition, for example when you call PluginCall with Function [PLUGIN_ONCE_A_SECOND](https://github.com/letscontrolit/ESPEasy/blob/ca10c834cc1e67f2b584dbdb833a67cc169464c5/src/__Plugin.ino#L1122-L1153), all the tasks and active and active ones are scanned and all the plugins are scanned to find the pointer to the funtion.

### My proposal is this:

- drop Plugin_id and Plugin_ptr
- Rename DeviceStruct like this:
```c
typedef boolean (*PlugIn_Handler)(byte function, struct EventStruct* event, String& args );

struct DevicePluginInfoStruct
{
  DevicePluginInfoStruct() :
    Added(false), Type(0), VType(0), Ports(0),
    PullUpOption(false), InverseLogicOption(false), FormulaOption(false),
    ValueCount(0), Custom(false), SendDataOption(false), GlobalSyncOption(false),
    TimerOption(false), TimerOptional(false), DecimalsOnly(false) {}  
  bool Added;
  byte Type;
  byte VType;
  byte Ports;
  boolean PullUpOption;
  boolean InverseLogicOption;
  boolean FormulaOption;
  byte ValueCount;
  boolean Custom;
  boolean SendDataOption;
  boolean GlobalSyncOption;
  boolean TimerOption;
  boolean TimerOptional;
  boolean DecimalsOnly;
  PlugIn_Handler Handler; // Replace  Plugin_ptr
  String DeviceName; 
  String[] ValuesName;
} DevicesPluginInfo[PLUGIN_MAX];

```
- Change ADDPLUGIN
```c
// Because of compiler-bug (multiline defines gives an error if file ending is CRLF) the define is striped to a single line
/*
#define ADDPLUGIN(NNN) \
  if (x < PLUGIN_MAX) \
  { \
    if (DevicesPluginInfo[x].Added == false)\
    {\
      DevicesPluginInfo[x].Added = true;\
      DevicesPluginInfo[x].DeviceName =  PLUGIN_NAME_##NNN; \
      DevicesPluginInfo[x].Handler = &Plugin_##NNN; \
    }\
    else\
    {\
       addLog(LOG_LEVEL_ERROR, String(F("PlugIn P")) + PLUGIN_##NNN + "-" + PLUGIN_NAME_##NNN +": There is another plugin named "+  DevicesPluginInfo[x].DeviceName  + " with the same id." ); +
    }\
    x++;\
  } \
  else \
    addLog(LOG_LEVEL_ERROR, FPSTR(ADDPLUGIN_ERROR));
*/
```
- removing ExtraTaskSettingsStruct
- removing all tasks information from SettingsStruct
- Create TaskInfoStrunct with the fields removed from SettingsStruct and fields of ExtraTaskSettingsStruct;
```c 
// Note: there are many fields that I do not know the function that perhaps could be removed if not used anymore.
struct TaskInfoStruct
{
/* Omitted ctor */
char[41]    Descripion;    //old ExtraTaskSettingsStruct.TaskDeviceName
char    Formula[VARS_PER_TASK][41];
char    ValueNames[VARS_PER_TASK][41];
long    PluginConfigLong[PLUGIN_EXTRACONFIGVAR_MAX];
byte    ValueDecimals[VARS_PER_TASK];
int16_t PluginConfig[PLUGIN_EXTRACONFIGVAR_MAX];
byte     PlugInID;  //old TaskDeviceNumber
union {
    struct {
      int8_t        Pin1;
      int8_t        Pin2;
      int8_t        Pin3;
      byte          Port;
    };
    int8_t        Pins[4];
  };
boolean       DevicePin1PullUp;
int16_t       PluginConfig[PLUGIN_CONFIGVAR_MAX];
boolean       Pin1Inversed;
float         PluginConfigFloat[PLUGIN_CONFIGFLOATVAR_MAX];
long          PluginConfigLong[PLUGIN_CONFIGLONGVAR_MAX];
boolean       OLD_SendData;
boolean       GlobalSync;
byte          DataFeed;
unsigned long Timer;
boolean       Enabled;
} Tasks[TASKS_MAX];
```
With this structure:

- to get all the info on plugins
```c
String pluginName = DevicesPluginInfo.[PLUGIN_023].DeviceName
....
```
- handle plugin function from task
```c
   DevicesPluginInfo.[Tasks[idtask].PlugInID].Handler(PLUGIN_ONCE_A_SECOND, event, arg);
....
```
- PluginCall with Function [PLUGIN_ONCE_A_SECOND] can transform in:
```c
...
        for (byte y = 0; y < TASKS_MAX; y++)
        {
          if(Tasks[y].Enabled && Tasks[y].DataFeed == 0 && Tasks[y].PlugInID>0)
          {
              DevicePluginInfoStruct pdi = DevicesPluginInfo.[Tasks[y].PlugInID];
              if (pdi.Added == true)
              {
                 if(pdi.Handler)
                 {
                   TempEvent.TaskIndex = y;                   
                   TempEvent.sensorType = pdi.VType;
                   TempEvent.OriginTaskIndex = event->TaskIndex; 
                   pdi.Handler(Function, &TempEvent, str);
                 }
                 else
                 {
                    //log invalid plugin function
                 }
              }
              else
              {
                 // Log error taks invalid pluginId
              }
          }
        }
....
```

### Expected behavior

view summary

### Actual behavior
view summary

Hardware: **All**

ESP Easy version: [mega-20180510](https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20180510)

TD-er, 2018-05-10:
There is even more redundant information stored in the settings.
For example the entire `DevicesPluginInfo[PLUGIN_MAX];` can be removed from memory, since it is all static information defined in the source or in the settings.

And I totally agree about the redundancy in the way to retrieve data.
It will lead to errors.
Even in the struct itself is redundant information.
For example `getValueCountFromSensorType(Device[DeviceIndex].VType)` should match `Device[DeviceIndex].ValueCount`.

But changing all you suggested will also change the way settings are stored.

workgroupengineering, 2018-05-10:
> For example the entire DevicesPluginInfo[PLUGIN_MAX]; can be removed from memory, since it is all static information defined in the source or in the settings.

I wouldn’t remove it because it is useful in the construction of the task ui.

>For example getValueCountFromSensorType(Device[DeviceIndex].VType) should match Device[DeviceIndex].ValueCount.
Only you know what is redundant. Are there any conde metric tools for Arduino? i can't find?

>But changing all you suggested will also change the way settings are stored.
Sure, a good option would be bson.