letscontrolit/ESPEasy #1397: No wifi in nodeMCU clone (no AP, no client)

HV-NL, 2018-05-10:
### Summarize of the problem/feature request

My nodeMCU clone does not have any wifi any more, using build mega-20180510

This seems the same problem as in item #1337 , so after solving it in build 20180502, the problem does now reappear.

### System configuration
nodeMCU / 4Mb / build mega-20180510 (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)

After upgrading wifi could not connect any more.
Instead of connecting I only get lots of `WIFI : Disconnected! Reason: '(201) No AP found'`

Then I flashed again, but now following a complete erase, but that did not solve it. Now I do not see any ESP AP, so cannot connect and configure.

Hardware: 'WeMos' nodeMCU clone (see #1337 for a picture of this device)

ESP Easy version: mega-20180510 (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)

### Extra info
I also flashed mega-20180510 into a Wemos D1 mini. 
Just like in item #1337, wifi works normally in this Wemos D1 mini.

ghtester, 2018-05-10:
How did you erase the flash? I would recommend to do it in accordance with this howto:
http://www.pratikpanda.com/completely-format-erase-esp8266-flash-memory/
Configure empty files to offsets 0x000000, 0x100000, 0x200000 and 0x300000 and press START to overwrite whole flash. Don't use ERASE button.

HV-NL, 2018-05-10:
I erased it by using the official python tool, as called by https://github.com/marcelstoer/nodemcu-pyflasher/releases

But erasing is not the relevant issue here.

TD-er, 2018-05-10:
Can you build the code from scratch? And then build it twice?
The thing is that yesterday I accidentally merged some pull request that was set to v2.0 (by accident) and thus touched the v2.0 branch.
This triggers the nightly build to build the v2.0 branch also.
I've noticed that changing between core 2.3 and 2.4 is almost a guarantee for broken builds. It can be fixed by building it twice.

HV-NL, 2018-05-10:
Yes, that did the trick. Now wifi is working again on my nodeMCU clone! This build indeed was broken.

closed 2018-05-10 by HV-NL

TD-er, 2018-05-10:
OK, then I will remove it from the releases list

svde, 2018-05-21:
ESPEasy_mega-20180519 is still broken.

TD-er, 2018-05-21:
@svde That's really strange, since nothing has changed there.
So it could have been one of those build issues.