letscontrolit/ESPEasy #1398: Cannot undo 'switch input - rotary encoder'

HV-NL, 2018-05-10:
### Summarize of the problem/feature request

After selecting 'switch input - rotary encoder' I cannot undo this, or even select another type of device (no pulldown menu is present any more).

### Steps to reproduce

1. take a fresh build 
2. configure task 1
3. use 'switch input - rotary encoder'

### System configuration
Hardware: Wemos D1 mini (but also tested with nodeMCU clone)

ESP Easy version: mega-20180510

ESP Easy settings/screenshots: 
[image]

workgroupengineering, 2018-05-10:
Press delete button

HV-NL, 2018-05-10:
I understand. Memory is a very precious item in the ESP system. I will close this question.

closed 2018-05-10 by HV-NL