letscontrolit/ESPEasy #1406: Switch & Pulse Plugin - GPIO select

TM4889, 2018-05-11:
Can anyone help me with the Switch and Pulse plugin? Where can i set the pin for this task? I found some pictures about this tasks where is a selectbox for this, but in the current version of ESPEasy is this not available.
[image]

Grovkillen, 2018-05-11:
By the looks of the GUI it seems that the version you're running is not the latest. Could you update and test again?

TD-er, 2018-05-11:
Could you list the version you are running?

TM4889, 2018-05-11:
I must do a Reset before i can use the current version 20180506, but there is no effect when i do this.

[image]

TD-er, 2018-05-11:
Could you try the 0507 build?
There is some issue with the nightly builds. Some work fine and some don't.
But the screenshot you made, was from a "working" version (at least it connected to wifi)
Can you get the information from the "more info" page? (button on the first page)

TM4889, 2018-05-11:
So have fix the problem with the reset and it is running now. I've create the switch task and set the GPIO, where my switch is connected to ground. Now i have the problem, that the state isn't changing

TM4889, 2018-05-11:
I have fix this problem too. It was a mistake by myself :)

closed 2018-05-11 by TM4889