letscontrolit/ESPEasy #1412: Bug on devices page

giig1967g, 2018-05-11:
Hi,
on devices page, in every task, only the first value is shown.

See image:

[image]

Grovkillen, 2018-05-11:
We have fixed this today. Should be working on next build.

Grovkillen, 2018-05-11:
Related to #1411  so I guess not entirely fixed.

micheldejongh, 2018-05-11:
It is not the case for all devices:
[image]

TD-er, 2018-05-11:
With the latest code ([committed today @9:40am](https://github.com/letscontrolit/ESPEasy/commit/06bc63e385b3fc252066f1d33965c4c2424aa6b8)) it should be fixed.
With the dummy device, you can select the output device type.
This is mainly for use with Domoticz.
I used the same sensor type value to determine the number of output values, but it appeared this sensor type thing was not consequent throughout the plugins. (See #1411 )
So I already changed it to use the ValueCount and not try to determine the number of output values.

uzi18, 2018-05-14:
@TD-er does automatic updates of devices values works for You? Have some reports about was working and from 10/11.05 doesn't.

TD-er, 2018-05-14:
Hmm last version on my node does update the device page.
I will load the latest source to see if something got broken :)

uzi18, 2018-05-14:
@TD-er ok thanks for this check.

Grovkillen, 2018-05-15:
Is this still a issue?

TD-er, 2018-05-15:
The original issue is fixed, but I'm not sure about @uzi18 's remark.
Maybe close this one and if @uzi18 did indeed find something that causes failure to load, then please add a new issue, preferably with the related (first and/or last received) JSON attached.

Grovkillen, 2018-05-15:
Deal.

closed 2018-05-15 by Grovkillen

uzi18, 2018-05-15:
Ok I will ask my friend for feedback.