letscontrolit/ESPEasy #1414: Web log does not work in MSIE11

ghtester, 2018-05-12:
### Summarize of the problem/feature request

Web log does not work in MSIE11 - The log window says only: Fetching log entries...

### Expected behavior

The Web log window should display log entries.

### Actual behavior

The log window says only: Fetching log entries...
In Firefox everything is OK.

### Steps to reproduce

1. Open http://ESP_EASY_IPaddr/log in MSIE 11
2. The log window says: Fetching log entries...
3. Nothing else is displayed even though Web log level is set to Debug More

### System configuration

Hardware: ESP12E

ESP Easy version: Official
20102  - Mega
Libraries
ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3
GIT version
mega-20180512
Plugins
72 [Normal] [Testing] [Development]
Build time
May 12 2018 02:14:56
Binary filename
firmware.bin

Grovkillen, 2018-05-12:
The fetch command that I implemented is not supported by IE11.

https://caniuse.com/#feat=fetch

Sorry but we need to ditch IE11 for the auto update of values and web log.

Future GUI will rely even more on that.

Grovkillen, 2018-05-12:
But I will add info about this in the GUI.

ghtester, 2018-05-12:
No problem, I expected that. ;-) But it would be nice if there's an info about unsupported (or partially supported) browsers. Thanks for your great work!

TD-er, 2018-05-12:
When looking for "Javascript fetch alternatives" I see a number of posts describing why alternatives are better.
For example [this post](http://jslancer.com/blog/2017/05/10/why-i-stop-using-fetch/)
Not sure if that helps IE11 support, or maybe the next command used is also unknown to IE11, but may be good to look at alternatives if they are better or easier to use (and result in smaller Javascript ;) )

Grovkillen, 2018-05-12:
Thanks for the link @TD-er

closed 2018-05-15 by Grovkillen