letscontrolit/ESPEasy #1415: Feature Request: Support for ESP board with integrated oled display

s0170071, 2018-05-12:
i just got myself one of these: 

https://www.ebay.de/itm/ESP8266-0-91-inch-OLED-Display-Screen-Nodemcu-WIFI-Board-CP2104-For-Arduino/

[image]

They are really small and they come with a 128 * 32 I2C OLED display already. 5,8€ with free shipping.
The display does not work out of the ESPEasy- Box. The seller gives this example sketch: 

 ```

#include <Wire.h>
#include "OLED.h"
//TTGO 0.91s OLED connection:
//SDA -- D4
//SCL -- D5
//RST -- D2
OLED display(SDA, SCL);
void setup() {
pinMode(D2, OUTPUT);
digitalWrite(D2, LOW); // turn D2 low to reset OLED
delay(50);
digitalWrite(D2, HIGH); // while OLED is running, must set D2 in high
Serial.begin(9600);
Serial.println("OLED test!");
// Initialize display
display.begin();
// Test message
display.print("Hello TTGO"); 
...
```

Any chance the current oled lib supports this already ? A little ?

TD-er, 2018-05-12:
I've seen a few of these already here or on the forum.
They have the reset wired and you should set the internal pull-up for the reset pin.

s0170071, 2018-05-12:
I've seen the display working. They have been delivered pre-programmed. So its not the reset. I don't even know which oled controller they use.

TD-er, 2018-05-12:
The default setting of ESPeasy is to disable the pull-up resistors.
Did you enable it on pin D2?
And make sure the SDA and SCL are set to the correct pins.

You should be able to use the standard OLED plugin (OLED framed does not yet support 128x32)

s0170071, 2018-05-12:
Right. 

D4 / D5 as SDA/SCL and you're good to go. I somehow assumed the SDA and SCL pins were fixed. 
Default setting is GPIO4 / GPIO5. That was obfuscating things a little. Thanks.

closed 2018-05-12 by s0170071