letscontrolit/ESPEasy #1419: Filename in binary firmware file may not exceed 48 bytes

TD-er, 2018-05-13:
From the crs2.py script:
`FILENAMEDUMMY = "ThisIsTheDummyPlaceHolderForTheBinaryFilename..." #48 chars`

The longest filename in our current release:
`ESP_Easy_mega-20180513_dev_ESP8266PUYA_1024_VCC.bin` is 51 bytes.

So my guess is that that file will probably get damaged by the python script.

We could decrease filename length, or increase the length of the dummy string to 64?

Can this be altered without any problems, or is there somewhere this length of 48 bytes hard coded?

s0170071, 2018-05-13:
The length of the dummy can be altered. The CRC routine is not doing anything with it.  Just check that it sits still in one of the two last sections of the binary. These two sections are not included in the CRC calculation.

Grovkillen, 2018-05-14:
@TD-er this is fixed right?

closed 2018-05-14 by Grovkillen