letscontrolit/ESPEasy #1425: Thingspeak controller use hostname does not work

s0170071, 2018-05-14:
on may 9th, all my sensors (different builds) stopped transmitting data to thingspeak. In the past this usually happened when thingspeak changed the IP address of their api.thingspeak.com domain. 
All my devices were set to use hostname. 

To overcome this issue a simple  reset does not work. I have to go to the controller settings and push submit once.  It appears that the dns lookup is done only once upon submit...

TD-er, 2018-05-14:
Huh, does this also happen on more recent builds?
I thought I fixed this a few months ago. At least it should be....

TD-er, 2018-05-15:
I will have a look at it.

TD-er, 2018-05-16:
I will have to look at the source code with this in mind.
I guess the IP it resolved to previously was still reachable but not accepting requests.

cbcalhoun, 2018-08-20:
Not sure if there is any progress on this issue. I also have noticed that the IP is cached in the controller config when using Thingspeak. Unfortunately they are receiving data into the AWS cloud so the IP is always changing. The fix is to disable and reenable the configured controller until the next time the IP is changed which is unreliable.

TD-er, 2018-08-23:
I found something regarding the return value of `connect` and applied that to this PR:
https://github.com/letscontrolit/ESPEasy/pull/1672
It may fix these issues also.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er