letscontrolit/ESPEasy #1426: no way to Right justify on display

jblb, 2018-05-14:
### Summarize of the problem/feature request
I was unable to right justify any value on display 

### Expected behavior
[System#Wifi_RSSI#R] should show value on right of display 

### Actual behavior
ERR is diplayed on right of display

### Steps to reproduce
put any value to display in Display - OLED SSD1306, if  i right justify with #R i got ERR displayedon right of the screen , if i remove the #R i got the value lef justifyed

seem that it comme from this commit
https://github.com/letscontrolit/ESPEasy/commit/5c3bf9e8944c0b490fd48c5518df0ba470052042

ESP Easy version: last git

jblb, 2018-05-14:
reading the code it was my thinking, but this give me a display lke that:

ERR-70
not right justifyed but the good value

giig1967g, 2018-05-14:
Ok. Will look at it and fix it tonight.

giig1967g, 2018-05-14:
PR created with #1427 
As a workaround please use:
 [System#Wifi_RSSI#VR]

TD-er, 2018-10-13:
I'll close this one now.
If it is still a bug, please re-open.

closed 2018-10-13 by TD-er