letscontrolit/ESPEasy #1428: Playground Plugin _P124_Ventus_W266_RFM69.ino update to work with current MEGA and integrate

sisamiwe, 2018-05-14:
Hi,
I'm using the _P124_Ventus_W266_RFM69.ino Plugin from Playground. It works fine.
Since the recent update of ESPEasy Mega, the Plugin does not work anymore since the GUI of that Plugin shows an error.

Please update the plugin to work the current MEGA and since the plugin works fine, please move it from playground to regular branch.

THX.

JustMe-NL, 2018-06-04:
I’ll look at the gui but the plugin needs a lot of RAM resources and the hardware is a bit to specific to be included in the standard build. I consulted with the maintainers and we concluded that it is best to keep this plugin as optional.

sisamiwe, 2018-06-04:
this is fine with me, but it would be great if the plugin can be updated to work with the current MEGA.

TD-er, 2018-06-04:
What's the difference between both plugins? (apart from one using a RF module)
Can they somehow be merged?

sisamiwe, 2018-06-04:
The P124 from Playground is using an RF-Module to directly listen to the RF signals from the outdoor station. There is no need for the indoor part. 
The other one is supporting the ESP connect to the indoor part.
But in my opinion they are more or less the same, since they getting the same data on different ways.

tonhuisman, 2022-04-18:
A specific Ventus build is available that supports this device, so I assume this issue can be closed.

closed 2022-04-18 by TD-er