letscontrolit/ESPEasy #1431: ESPeasy and IDX on JSON string

pitrio, 2018-05-15:
### Summarize of the problem/feature request

dear all,
I would like add a IDX number (used to link with Domoticz) on ESPeasy JSON string, in particulary in the "sensors" position.
I saw that the JSON string are handled on webserver.ino at "void handle_json()" routine.

Also we can to add all IDX value, becouse we can put up to 3 IDXs code.

thank you!

### Expected behavior

"Sensors":[
{
"TaskValues": [
{"ValueNumber":1,
"Name":"Analog",
"NrDecimals":2,
"Value":9.00
}],
"TaskNumber":1,
"TaskEnabled":"true",
"TaskInterval":1,
"Type":"Analog input - internal",
"TaskName":"ADC 1",
**"IDX":"8"**
},

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2018-05-15:
If you have multiple controllers, then there are multiple idx values possible.
So, what to do then?
Is "IDX" an array/list ?

Grovkillen, 2018-05-15:
If we decide to have hard coded suffix numbers it will make my JavaScript have to cope with more possible scenarios. If we could have them connected to task value number it would be alot better. I.e.:

```
{
"Sensors":[
{
"TaskValues": [
{"ValueNumber":1,
"Name":"",
"NrDecimals":2,
"Value":0.00
}],
"TaskNumber":1,
"TaskEnabled":"false",
"TaskInterval":0,
"Type":"Gases - CO2 Senseair",
"TaskName":"",
"IDX":[{"Value":123, "Number":1},{"Value":456, "Number":2},{"Value":678, "Number":3}]
},
{
"TaskValues": [
{"ValueNumber":1,
"Name":"Analog",
"NrDecimals":2,
"Value":0.00
}],
"TaskNumber":2,
"TaskEnabled":"false",
"TaskInterval":0,
"Type":"Analog input - PCF8591",
"TaskName":"",
"IDX":[{"Value":123, "Number":1},{"Value":456, "Number":2},{"Value":678, "Number":3}]
},
{
"TaskValues": [
{"ValueNumber":1,
"Name":"Temperature",
"NrDecimals":2,
"Value":0.00
},
{"ValueNumber":2,
"Name":"Humidity",
"NrDecimals":2,
"Value":0.00
},
{"ValueNumber":3,
"Name":"Pressure",
"NrDecimals":2,
"Value":0.00
}],
"TaskNumber":3,
"TaskEnabled":"false",
"TaskInterval":60,
"Type":"Environment - BMx280",
"TaskName":"",
"IDX":[{"Value":123, "Number":1},{"Value":456, "Number":2},{"Value":678, "Number":3}]
}
],
"TTL":60000
}
```

This way we are able to parse through the string without needing to know what objects that are in there. We will know that the number of IDX:s are 3 by just looking at the IDX object. If we have multiple POSSIBLE object that we need to look at we would need to have the script itself know that these IDX1, IDX2 and IDX3 are related even though they are programmatically not related.

On the note of missing IDX input. This is a screenshot of todays release:

[image]

TD-er, 2018-05-15:
You have to configure a controller which can handle idx values for the text box to appear at that index.
I guess adding an array/list is better. Then we can also add the enabled status.

Grovkillen, 2018-05-15:
Then we should not name it "IDX" but "Data Acquisition"

```
"DataAcquisition":
[{
"Controller": 1,
"IDX":123,
"Endabled": false}{
"Controller": 2,
"IDX":111,
"Endabled": true}
"Controller": 3,
"IDX":222,
"Endabled": true}
}]
```

Grovkillen, 2018-05-15:
And if no IDX is used we just skip that object for that controller.

pitrio, 2018-05-16:
dear all,
good idea "Data Acquisition" field!

I saw in the new version ... the last IDX input is ok... 

thank you.

Grovkillen, 2018-05-17:
Implemented

closed 2018-05-17 by Grovkillen