letscontrolit/ESPEasy #1445: [JSON] Make state available via JSON

TD-er, 2018-05-17:
We have static data (settings stored in flash) and dynamic data.
The dynamic data consists of:

- State set via commands and/or web interface.
- (future) recorded data
- sensor data. (already available via `/json` url )

This issue is about the state.
We need a new call to plugins supporting commands to report their state.
This should be reported back via JSON.

A request should be addressable to either any task, or a specific.
Perhaps also a selection based on support for a command should be supported? (e.g. show dimmed state of any display)

workgroupengineering, 2018-05-17:
instead of using the URL to determine the format type, I would use http content-type hader:

Example if you get the url http: //192.168.1.X/devices with Content-Type: text / html will remain html with Content-Type: application / json the json, Content-Type: application / bson in bson , etc...

TD-er, 2018-05-17:
Only thing is we have to write all encoding ourselves.
Also setting the content type in the request is a bit harder.

The general idea (call it a vision, planning, etc.) is to start using JSON as communication form between platforms to exchange data.
So HTML will be too specific and if someone wants to have a html format, they can generate it based on the JSON.