letscontrolit/ESPEasy #1446: [JSON] Content-Type negotiation

workgroupengineering, 2018-05-17:
### Summarize of the problem/feature request
when calling an url es: http: //192.168.1.X/devices return the content based on the html html Content-Type

### Expected behavior

Example if you get the url http: //192.168.1.X/devices with Content-Type: text / html will remain html with Content-Type: application / json the json, Content-Type: application / bson in bson , etc...

### Actual behavior
none

Hardware:

ESP Easy version: **all**

Related issue: #1443 #1445 
### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2018-05-17:
How to set the content-type in the request?
And what content-types should be supported?

My idea was to leave the current implementation in tact and add some new -more generic- ways to interact with:

- static data (settings, stored in flash)
- recorded data (semi-static data, may be stored in SPIFF or elsewhere)
- state (semi-static data, not stored in flash)
- dynamic data (sensor data)

This is all content and thus should be in a simple form.
Then there is the model + view, which can be a set of html/css templates + JavaScript (or whatever browser handled language).

I think that is a simpler approach and will result in more clean code, simpler tasks and less use of resources on the ESP.

Following your suggestion should give the same data, but just in another form.
The current `/devices` URL is a representation of the view + edit of the devices (plugins/tasks).
I think the user will expect that behavior and if we just return that in a JSON form, there is some expectation to be able to edit the values.

I agree on making sure to have a proper tree structure to serve the data and it is matter of taste how to set the resulting output format.
But I don't want to change the current webpage structure to allow for different formats of the same data, simply because the current structure is too complex in some cases.
It is now based on the current structs to keep the data and those are perfectly defend-able given how the project started, but given the current state and direction of the project are now sub-optimal.
So I do not want to continue making interfaces based on those structures, but try to make the interface more clean.

workgroupengineering, 2018-05-17:
>How to set the content-type in the request?

Yes

>And what content-types should be supported?

For sure: application / json the json, application / bson, text / xml. If you have an intention to use cereal, most of these would be easy to implement.

> My idea was to leave the current implementation in tact and add some new -more generic- ways to >interact with:
>
>   static data (settings, stored in flash)
>    recorded data (semi-static data, may be [...]

>This is all content and thus should be in a simple form.
>Then there is the model + view, which can be a set of html/css templates + JavaScript (or whatever >browser handled language).

>I think that is a simpler approach and will result in more clean code, simpler tasks and less use of >resources on the ESP.

With the approach suggested by me the html forms could be a templete saved in gzip format on the flash (WebStaticData.h) and populated through javascript that queries the webserver with content-type json.
the webserver and plugins code would be cleaner and more efficient.

> Following your suggestion should give the same data, but just in another form.
> The current /devices URL is a representation of the view + edit of the devices (plugins/tasks).
> I think the user will [...]

 If you use the [REST](https://en.wikipedia.org/wiki/Representational_state_transfer) + [Hypermedia ](https://en.wikipedia.org/wiki/HATEOAS) approach it is very easy to implement mourning.

closed 2020-12-30 by workgroupengineering