letscontrolit/ESPEasy #1447: Disconnected! Reason: '(8) Assoc leave' Connected for 12 s

s0170071, 2018-05-17:
What is this Assoc leave? This goes on forever on one of my recently updated units. 
I am still in the process of finding out how far back this goes...

```
INIT : Booting version: (custom) (ESP82xx Core 41a64707, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
86 : INIT : Warm boot #4
87 : FS   : Mounting...
112 : FS   : Mount successful, used 497231 bytes of 957314
127 : CRC  : No program memory checksum found. Check output of crc2.py
152 : CRC  : SecuritySettings CRC   ...OK 
259 : INIT : Free RAM:27136
259 : INIT : I2C
259 : INIT : SPI Init (without CS)
1595 : INFO : Plugins: 49 [Normal] (ESP82xx Core 41a64707, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
1595 : WIFI : Set WiFi to STA
1627 : WIFI : Connecting MNET attempt #0
1747 : WD   : Uptime 0 ConnectFailures 0 FreeMem 26352
2095 : SYS  : 26248.00
2126 : HTTP : connection failed
2129 : SYS  : 0.00
2159 : HTTP : connection failed
5464 : WIFI : Connected! AP: MNET (5C:49:79:28:0C:6F) Ch: 6 Duration: 3836 ms
7286 : WIFI : DHCP IP: 192.168.1.25 (Garage-3) GW: 192.168.1.1 SN: 255.255.255.0   duration: 1822 ms
10669 : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 5205 ms
10912 : WIFI : Connecting MNET attempt #0
12273 : WIFI : Connected! AP: MNET (5C:49:79:28:0C:6F) Ch: 6 Duration: 1360 ms
15148 : WIFI : DHCP IP: 192.168.1.25 (Garage-3) GW: 192.168.1.1 SN: 255.255.255.0   duration: 2875 ms
18559 : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 6285 ms
18568 : WIFI : Connecting MNET attempt #0
19210 : WIFI : Connected! AP: MNET (5C:49:79:28:0C:6F) Ch: 6 Duration: 641 ms
20826 : WIFI : DHCP IP: 192.168.1.25 (Garage-3) GW: 192.168.1.1 SN: 255.255.255.0   duration: 1615 ms
24182 : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 4972 ms
24188 : WIFI : Connecting MNET attempt #0
24519 : WIFI : Connected! AP: MNET (5C:49:79:28:0C:6F) Ch: 6 Duration: 330 ms
26374 : WIFI : DHCP IP: 192.168.1.25 (Garage-3) GW: 192.168.1.1 SN: 255.255.255.0   duration: 1855 ms
29757 : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 5239 ms
29764 : WIFI : Connecting MNET attempt #0
30245 : WIFI : Connected! AP: MNET (5C:49:79:28:0C:6F) Ch: 6 Duration: 479 ms
31747 : WD   : Uptime 1 ConnectFailures 2 FreeMem 23096
32186 : WIFI : DHCP IP: 192.168.1.25 (Garage-3) GW: 192.168.1.1 SN: 255.255.255.0   duration: 1940 ms
35531 : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 5286 ms
35538 : WIFI : Connecting MNET attempt #0
35858 : WIFI : Connected! AP: MNET (5C:49:79:28:0C:6F) Ch: 6 Duration: 319 ms
37741 : WIFI : DHCP IP: 192.168.1.25 (Garage-3) GW: 192.168.1.1 SN: 255.255.255.0   duration: 1883 ms
41082 : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 5224 ms
41324 : WIFI : Connecting MNET attempt #0
41809 : WIFI : Connected! AP: MNET (5C:49:79:28:0C:6F) Ch: 6 Duration: 483 ms
43345 : WIFI : DHCP IP: 192.168.1.25 (Garage-3) GW: 192.168.1.1 SN: 255.255.255.0   duration: 1536 ms
46731 : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 4922 ms
46738 : WIFI : Connecting MNET attempt #0
47667 : WIFI : Connected! AP: MNET (5C:49:79:28:0C:6F) Ch: 6 Duration: 928 ms
49496 : WIFI : DHCP IP: 192.168.1.25 (Garage-3) GW: 192.168.1.1 SN: 255.255.255.0   duration: 1829 ms
52857 : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 5189 ms
53099 : WIFI : Connecting MNET attempt #0
54047 : WIFI : Connected! AP: MNET (5C:49:79:28:0C:6F) Ch: 6 Duration: 947 ms
55650 : WIFI : DHCP IP: 192.168.1.25 (Garage-3) GW: 192.168.1.1 SN: 255.255.255.0   duration: 1603 ms
66645 : WIFI : Connecting MNET attempt #0
66651 : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 12 s
67121 : WIFI : Connected! AP: MNET (5C:49:79:28:0C:6F) Ch: 6 Duration: 475 ms
68968 : WIFI : DHCP IP: 192.168.1.25 (Garage-3) GW: 192.168.1.1 SN: 255.255.255.0   duration: 1847 ms
72304 : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 5183 ms
72546 : WIFI : Connecting MNET attempt #0
73478 : WIFI : Connected! AP: MNET (5C:49:79:28:0C:6F) Ch: 6 Duration: 930 ms
75122 : WIFI : DHCP IP: 192.168.1.25 (Garage-3) GW: 192.168.1.1 SN: 255.255.255.0   duration: 1644 ms
78530 : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 5051 ms
78537 : WIFI : Connecting MNET attempt #0
79475 : WIFI : Connected! AP: MNET (5C:49:79:28:0C:6F) Ch: 6 Duration: 937 ms

```

Grovkillen, 2018-05-17:
https://www.letscontrolit.com/wiki/index.php/Wifi

> 8
> Disassociated because sending STA is leaving (or has left) BSS
> Operating System moved the client to another access point using non-aggressive load balancing.

burkulesomesh43, 2019-05-26:
@Grovkillen Can you please explain in detail non aggrasive load balancing?
I am also facing same issue related to wifi. getting disconnect reason ASSOC_LEAVE(8).

Grovkillen, 2019-05-26:
To my knowledge the non-aggressive load balancing is that it the AP is busy and the client is trying another AP. But I'm no expert.

TD-er, 2019-05-26:
Aggressive Load Balancing and non-aggressive load balancing are indications the selected AP is responding with a "busy" reply when you try to connect a client.
The difference is that with the non-aggressive variant the client will finally get a successful connection after some number of retries.

So, is your AP busy?
Is it still a problem after rebooting the AP?
It is quite common for a client to get disconnected from the AP. It may become an issue when it is hardly possible to reconnect.

KamensekD, 2020-07-13:
My ESP8266 D1 Mini drove me crazy by disconnecting from my home wifi about every minute with message: Disconnected! Reason: '(8) Assoc leave'
No ESPEasy configuration changes made any difference. After few days I finally found the problem - in my case it was DHCP server client lease time on my Linksys router, which was configured to 1 minute! Changed it to 0 (which means one day) and disconnects stopped :)

TD-er, 2020-07-13:
That's also good to know.
I am looking into the (other) WiFi issues right now, so I will try to keep in mind to add a warning if the DHCP TTL time is too low to be practical.

kriger1965, 2023-11-17:
... i receogniced this ... with hall my other devices i am in the same network without issues
- i reassigned my dhcp servoer for a long term AdressTime ...more than 86400 or set to 0
- also i tryed to give the ems esp a fixed ip 192.168.1.120 - so not needed to contact the dhcp service

kriger1965, 2023-11-17:
but sometimes it display´s: 
[image]

tonhuisman, 2023-11-17:
> but sometimes it display´s: [image]

This doesn't look like ESPEasy to me... 🤔

kriger1965, 2023-11-17:
Yes it is still a resistent problem after rebooting the device and the access point

kriger1965, 2023-11-17:
Sorry - i mixed up the tasks of the githup - chat ... i am in troubleshooting with ems-esm  Board ...