letscontrolit/ESPEasy #1455: timer1 fires every 30 seconds repeatedly

artua, 2018-05-21:
SELF COMPILE for minimum binary size (only few modules active)

### Summarize of the problem/feature request
timer1 if set once, fires repeatedly every 30 seconds. even timerSet1,0 did not work.
timer2 has no such issue

### Expected behavior
Timer fires only once

### Actual behavior
Timer fires every 30 seconds

### Steps to reproduce
``` Set in rules:
on SW1#SW=1 do
  timerSet,1,1
  Publish %sysname%/inside,timer1_started
endon

on rules#timer=1 do
Publish %sysname%/inside,timer1_event
endon
```

Unit restarted, problem persists.
### System configuration
Hardware: Sonoff T1 UK 2G

ESP Easy version: 
commit 147acc672486caa6d0ce71f5a242cc6061113cc0 (tag: mega-20180519, origin/mega, origin/HEAD)
Author: ESPEasy release bot <edwin@datux.nl>
Date:   Sat May 19 04:00:07 2018 +0200

### Rules or log data
```
5/21/2018, 11:46:57 AMnode: 9d68205e.c5bcf
L_Kitchen/in : msg.payload : string[60]
"{"idx":1,"RSSI":9,"command":"switchlight","switchcmd":"Off"}"
5/21/2018, 11:46:58 AMnode: 9d68205e.c5bcf
L_Kitchen/inside : msg.payload : string[14]
"timer1_started"
5/21/2018, 11:46:58 AMnode: 9d68205e.c5bcf
L_Kitchen/inside : msg.payload : string[12]
"timer1_event"
5/21/2018, 11:47:30 AMnode: 9d68205e.c5bcf
L_Kitchen/inside : msg.payload : string[12]
"timer1_event"
5/21/2018, 11:48:01 AMnode: 9d68205e.c5bcf
L_Kitchen/inside : msg.payload : string[12]
"timer1_event"
5/21/2018, 11:48:32 AMnode: 9d68205e.c5bcf
L_Kitchen/inside : msg.payload : string[12]
"timer1_event"
```

v-a-d-e-r, 2018-05-21:
Tested here, and the timer is only triggered 1 time as long as the button state is '1'. No issue....

artua, 2018-05-21:
Will flash official build today to test it

workgroupengineering, 2018-05-21:
it's a design behavior, if you want to restart the timer you call timerSet again.
```
on rules#timer=1 do
  Publish %sysname%/inside,timer1_event
  timerSet,1,30
endon
```

artua, 2018-05-21:
I never requested that timer again. It fires continuously every 30 seconds after first use.

workgroupengineering, 2018-05-21:
the timer after it has elapsed is disabled. To restart it, as I have already said, you call timerSet,1,30 again.

```
on rules#timer=1 do
  Publish %sysname%/inside,timer1_event
  timerSet,1,30
endon
```

TD-er, 2018-05-21:
@workgroupengineering I guess you both are talking about something different.
@artua claims the timer is being called every time, without being called.
His MQTT message log shows only one timer-set event and multiple timer1 events.

Did this work on previous versions?

Grovkillen, 2018-05-31:
We need more info.

artua, 2018-06-20:
Hello,

found an error in rules file. Please close an issue.

closed 2018-06-20 by artua

TD-er, 2018-06-20:
Could you please post the correct rule (or at least some indication to what went wrong) here, so others may be helped when searching for similar issues.

artua, 2018-06-20:
It was totally my mistake, sorry to disturb the community.