letscontrolit/ESPEasy #1458: uptime is less than WI-FI connection time

workgroupengineering, 2018-05-22:
### Summarize of the problem/feature request

the same as the title

### Expected behavior

Show correct connection time

### Actual behavior

the same as the title

### System configuration

Hardware: **Sonoff TH10/16**

ESP Easy version: 
ESP Easy | Information |
-----|-----|
Build|20102  - Mega|
Libraries|ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3|
GIT version|mega-20180518|
Plugins|72 [Normal] [Testing] [Development]|
Build time|May 18 2018 02:19:33|
Binary filename|ESP_Easy_mega-20180518_dev_ESP8266_1024.bin|

ESP Easy settings/screenshots:

Budman1758, 2018-05-22:
Is it possible it was on a different channel for the time difference?

TD-er, 2018-05-22:
It is possible to get this difference.
The connected time is based on `millis()` and the uptime is based on the system clock, which can be corrected via NTP.
After 24.x days there will be a difference also.
But I guess nobody had 24 days connection yet ;)

Grovkillen, 2018-05-31:
I will close this issue.

closed 2018-05-31 by Grovkillen