letscontrolit/ESPEasy #1462: PCA9685 plugin - control via MQTT not possible?

micropet, 2018-05-24:
By very brief information about the PCA9685 plugin, I understand that the PWM settings only work through the web interface.

The wiki says, "You can use the PWM levels using the http commands to ESP Easy."

In order to be able to work with it in real operation, control via MQTT is important.

Grovkillen, 2018-05-24:
You can use the same commands through serial or MQTT. Most commands are
protocol agnostic.

Grovkillen, 2018-05-24:
https://www.letscontrolit.com/wiki/index.php/ESPEasy_Command_Reference

and the wiki page for that device says: `PCAPWM,<pin>,<value>`

So the correct topic: `ESP-201/cmd`
And the command: `PCAPWM,1,2000`

closed 2018-05-24 by Grovkillen

micropet, 2018-05-24:
you are right:
mosquitto_pub -t ESP-201/cmd -m 'PCA9685.PCAPWM 1,2000'
and 
mosquitto_pub -t ESP-201/cmd -m 'PCA9685.pwm 1,2000'
seems to work.

I am not at home and only connected to my computer via RDP.
So I can not watch the ports.

OK, then I'll close it now. Many Thanks.

micropet, 2018-05-24:
oh, just tested:

 mosquitto_pub -t ESP-201/cmd -m 'pwm 2,4000'

and got in the Log:
10041624: SW : GPIO 2 Set PWM to 4000