letscontrolit/ESPEasy #1464: RSSI - different values.

melwinek, 2018-05-24:
I noticed something strange in the value of RSSI in Sonoff S20.
In ESP12F it's good.
Details in the attachment.
Two different values.
Version 20180522 compiled in platformio.
[image]
[image]

v-a-d-e-r, 2018-05-24:
With Sonoff Basic nothing "strange" to see. All values are the same.
(RSSI):	802.11N (-74 dB)
System info	RSSI: -74.00

TD-er, 2018-05-25:
The values in the device page are dynamically refreshed and the one on the first page is not.

Grovkillen, 2018-05-25:
The upcoming GUI will take care of all this, everything will be dynamic then.

melwinek, 2018-05-25:
This is not a refresh problem.
In this device, both values are totally different.
In devices is never below 80.
In main never exceeds 70.
I will try to flash the official build.

melwinek, 2018-05-25:
@Grovkillen Change switches/relays state would also be made by clicking?
That would be interesting.

Grovkillen, 2018-05-25:
Yep that's on the list to

Grovkillen, 2018-06-04:
Should we close this? Cannot reproduce...

melwinek, 2018-06-06:
You can close.
This is probably something with the hardware, on another sonoff is ok.

closed 2018-06-06 by Grovkillen