letscontrolit/ESPEasy #1467: Question: multiple tasks with same name used in rules

clumsy-stefan, 2018-05-25:
Not sure if this is intentionally or a bug, but if I have multiple tasks with the same name but different values (task1: weather / val1: temp, task2: weather / val1: hum, task3: weather / val1: press), I can only use the task1/val1 pair in rules. if I use "if [weather#hum]>10" it never matches.

I have this situation multiple times, as I don't want sepparate task names for the "same" logic gruop (eg. different environment sensors in a weather station, button and relays in a switch, etc) because it generates sepparate devices on my server for each different task-name..

Is this behaviour wanted or just not yet sought of? Any comments welcome!

TD-er, 2018-05-25:
Current behavior is to have unique combinations, so task names should be unique.
The task name is used to find a task number and if found, it will stop looking.
So you cannot have a "foreach" like behavior.

clumsy-stefan, 2018-05-25:
ok, thanks for the quick reply...

so then just a suggestion from me would be to make the comination task/value unique... which would solve this...

btw: while playing with this I also discovered that certain commandsa (eg. setinputswitchstate) use task-numbers starting with 0, others (like TaskRun) start at 1 (as in the web-display)... probably also intentionally, but somtimes irritating...

clumsy-stefan, 2018-05-25:
sure... probably there should be some guidelines for developers ;) especially such as myself who basically code without any rules 😬

TD-er, 2018-05-25:
If you have a list of these, we can at least document it, or try to make it more uniform.

Grovkillen, 2018-05-25:
@TD-er you asking me?

TD-er, 2018-05-25:
Nope, it wasn't directed at someone in particular, but @clumsy-stefan found it, so maybe he noticed some others already that were behaving inconsistently.
So better keep this knowledge than acknowledge its inconsistency and disregard this information :)

Grovkillen, 2018-05-25:
True! Agree

clumsy-stefan, 2018-05-25:
i'll keep an eye on it too.... I mean it's nothing critical, but if the module developer sees this he can probably act.... if he likes...

Grovkillen, 2018-05-31:
Is this something we need to keep open or can I close it?

clumsy-stefan, 2018-05-31:
For me I'd favor a change which would take the task name/value name pair as uniqe identifier so we can have multiple tasks under the same name with different values... But I guess it's up to the main developpers to decide if such a thing would be usefull or not!

therwise from my point of view this can be closed!

thanks!