letscontrolit/ESPEasy #1472: Ignore DS18B20 85°C power-on reset value

andrea-daru, 2018-05-29:
When the sensor has any kind of problem (disturb on the line, conversion not completed...) return +85°C value.
I want to modify the firmware to manage the error ignoring the reading equal to 85.

The same thing's it's done in OpenHab2 for directly connected 1wire devices
[https://docs.openhab.org/addons/bindings/onewire1/readme.html](url)
> if **ignore85CPowerOnResetValues** is included, the power-on reset value (+85°C) of DS18B20 devices is ignored.

Do you think that there's a better solution?

Grovkillen, 2018-05-29:
https://github.com/letscontrolit/ESPEasy/pull/1245

TD-er, 2018-05-29:
The returned value is now a NaN when there is some kind of invalid value.
This value should not be sent to a controller.
If an invalid value is sent, please give more information on the used configuration.