letscontrolit/ESPEasy #1474: PWM function ==> unknown command

DittelHome, 2018-05-31:
### Summarize of the problem/feature request
Since ESP_Easy_mega-20180524_normal_ESP8266_4096 pwm is not working anymore.
In response to pwm,5,800 you get only a "Unknown command!"
With ESP_Easy_mega-20180522_normal_ESP8266_4096 all is working fine..

### Expected behavior
After typing pwm,5,800 the esp should set gpio 5 to 800 
```
{
"log": "GPIO 5 Set PWM to 800",
"plugin": 1,
"pin": 5,
"mode": "PWM",
"state": 800
}

```

### Steps to reproduce
type pwm,5,800 in gui (tools command)

### System configuration
Hardware:
ESP mini V1

Grovkillen, 2018-05-31:
I can confirm this.

TD-er, 2018-05-31:
Very likely is this a duplicate of this issue: #1471

DittelHome, 2018-05-31:
#1471 concerns  GPIO commands.
Problem here is PWM command.

TD-er, 2018-05-31:
@DittelHome That is correct, but it looks like the GPIO command itself is not the cause in that issue.
The command match for P022 only matches "name.gpio" kind of string.
So there must be another reason why the command handling has changed.
And because of that uncertainty, I did not yet change/revert that commit.
We have to look further into these 2 issues.

workgroupengineering, 2018-06-01:
@DittelHome see [this](https://github.com/letscontrolit/ESPEasy/issues/1471#issuecomment-393686128).

closed 2018-12-02 by DittelHome