letscontrolit/ESPEasy #1476: Crash when using taskRun with Nextion

waspie, 2018-05-31:
### Summarize of the problem/feature request
I'm using a Nextion to create a thermostatic set point by way of dummy values and taskRuns to update the display

### Expected behavior
you press up, it adds to the setpoint and taskRun runs the dummy to update the display.

### Actual behavior
it happens, then crashes

12310 : [if 60>=60 and 60<81]=true
12314 : LoadFromFile: config.dat index: 6144 datasize: 472
12319 : LoadFromFile: config.dat index: 4096 datasize: 472
12323 : ACT  : TaskValueSet,2,1,60+1
12335 : Command: taskvalueset
12339 : ACT  : delay 200
12347 : Command: delay
12549 : ACT  : taskRun,2
12557 : Command: taskrun
12559 : LoadFromFile: config.dat index: 6144 datasize: 472
12562 : Dummy: value 1: 61.00
12562 : Dummy: value 2: 60.00
12562 : Dummy: value 3: 0.00
12562 : Dummy: value 4: 0.00
12566 : EVENT: atticBed#setPoint=61.00
12592 : ACT  : nextion,page1.t0.txt="61.00"
page1.t0.txt="61.00"

Panic /home/travis/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:99 __yield

ctx: cont 
sp: 3fff3250 end: 3fff415c offset: 01b0

### Steps to reproduce
Add a nextion and dummy device

yes
### System configuration

[image]
[image]
[image]

Hardware: wemos d1

ESP Easy version:  mega testing from may 24 2018

ESP Easy settings/screenshots: 

### Rules or log data

```
on lrNextion#code=258.00 do
if [atticBed#setPoint]>=60 and [atticBed#setPoint]<81
TaskValueSet,2,1,[atticBed#setPoint]+1
endif
taskRun,2
endon

on lrNextion#code=259.00 do
if [atticBed#setPoint]<=80 and [atticBed#setPoint]>60
TaskValueSet,2,1,[atticBed#setPoint]-1
taskRun,2
endif
endon

on atticBed#setPoint do
nextion,page1.t0.txt="%eventvalue%"
endon
```

TD-er, 2018-05-31:
Such a yield() panic is normally occurring only when something is calling `delay()` or `yield()` from within a function that has not been called from either `setup()` or `loop()`.
That can only be from a function which is a call-back from something like an eventhandler.
At this moment I don't see in the code what could be causing that in the P075 source code.

Another thing that strikes me as odd, is the amount of work being done in the ten_per_second  case.
I would suspect the most time critical is to make sure all serial data is collected and read the buffers in time. But a lot more is done, which could take more than 100 msec to process.

waspie, 2018-05-31:
that all sounds well above my pay grade. I replaced the taskRun with
"event,atticBed#setPoint=[atticBed#setPoint]"

and that's triggering the display update event in the meantime. That'll be fine for my use but i brought this up in case it's something that needs fixed anyway! thanks

waspie, 2018-06-01:
how do i self compile with the nextion plugin (75) I leave it in the directory with the rest like i always do but i never get it as an option in devices.
there's no ifdef=testing stuff in it so i'm at a loss?

waspie, 2018-06-01:
the more i mess around with it the more it seems centered around calling an event. if i do everything by ifs and set them all inside the ifs (making my rules bigger) it all works but when i try to call an event to update the display that's when it craps out.

waspie, 2018-06-10:
real life example:

`
on livingRoom#bt do
event,button[livingRoom#bt]
endon

on button258 do
if [page2#light1]=0
Publish /%sysname%/light1,1
nextion,page1.va0.val=1
TaskValueSet,4,1,1
else
Publish /%sysname%/light1,0
nextion,page1.va0.val=0
TaskValueSet,4,1,0
endif
endon

on lrPanel#lrl1 do
if [lrPanel#lrl1]=1
TaskValueSet,4,1,1
else
TaskValueSet,4,1,0
endif
event,button258
endon
`
results in:
`
12788 : EVENT: livingRoom#bt=258.00
12796 : ACT  : event,button258
12800 : Command: event
12800 : EVENT: button258
12817 : ACT  : Publish /lrNextion35/light1,0
12820 : Command: publish
12822 : ACT  : nextion,page1.va0.val=0
page1.va0.val=0

Panic C:\Users\waspi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\core_esp8266_main.cpp:99 __yield

ctx: cont 
sp: 3fff1dc0 end: 7fffffff offset: 01b0

>>>stack>>>
3fff1f70:  00000001 00000001 3fff206c 40232430
(lots more numbers/letters here)
`

giig1967g, 2018-06-11:
Hi,
I have the same problem using switch plugin. So I think that the issue is not plugin related.
In my case if I call an event and inside this event there is a taskrun the unit crashes.
i.e.:

```
on p1#switch do
  [some code]
  event,event1
endon

on event1 do
  [some code]
  taskrun,1
endon
```

My workaround is to create a timer:
```
on rules#timer=1 do
  taskrun,1
endon

on p1#switch do
  [some code]
  event,event1
endon

on event1 do
  [some code]
  timerset,1,1
endon
```

It looks like a "yield" issue...

TD-er, 2018-06-11:
The error suggests a `yield()` or `delay()` is being called from within a call-back function. (or event handling)
Callback functions should only set a flag and the normal loop should then pick it up and act upon it.

giig1967g, 2018-06-11:
agreed, but how can it be fixed?

TD-er, 2018-06-11:
Not sure yet, we need to look into it to get a better idea.
But your information may suggest it is not really an issue with the Nextion plugin, but that it apparently manifests itself there in a more reproducible (or noticeable) way.

giig1967g, 2018-06-11:
correct.
With the workaround it works nicely. So the problem is the time it takes to evaluate the command taskrun within the event call.

giig1967g, 2018-06-19:
Hi,
what is current situation of this analisys?
As the problem is not the TaskRun command itself, but the way the interpreter takes to execute the commands, it could hide something bigger than it seems.

TD-er, 2018-06-19:
I've not had the time past week to look into it.
(today will also be quite busy)

I really need some way to organize all thoughts, issues and to-do's about this project.
If anyone has some idea how to get-things-done, I'm all ears. Github Issues are hard to sort and arrange into some useful tasks to do in one evening.

closed 2022-04-21 by TD-er