letscontrolit/ESPEasy #1477: Moving out gpio, pwm, from P001 Switch

workgroupengineering, 2018-06-01:
### Summarize of the problem/feature request

After #1448, there is  a bug on execution of commands gpio, pwm, ecc... (#1471 , #1474 ) , because with the commit #1448 PluginCall is executed only active tasks and if P001 Switch is not in the list of devices, the commands gpio and other are not executed.

### Expected behavior

gpio and other commads are executed.

### Actual behavior

if P001 Switch is not in the list of devices, the commands gpio and other are not executed.

### Steps to reproduce

1. do not add the plugin to the list of devices
2. Call command gpio from web or other

### System configuration

Hardware: **all**

ESP Easy version: [mega-20180524](https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20180524)

workgroupengineering, 2018-06-01:
I have identified two possible solutions.
1. move the gpio command, etc. into Command.ino ExecuteCommand
2. Create the 000_Core plugin that is always executed.

what is the best choice?

TD-er, 2018-06-01:
If there is a command that's not related to a specific plugin, then it should not be in the plugins.
The "GPIO" command you added for P022, has a prefix, right? So it should not conflict with the general "GPIO" command used to set any pin not connected to an expander.

waspie, 2018-06-03:
What he's saying is, the GPIO command isn't available UNLESS the switch plugin is used for any reason.

There are many occasions where I don't need a "switch" device but still use gpio commands. 

NO switch device:

gpio,5,0
>gpio,5,0
380499 : Command: gpio
380500 : Command unknown

Unknown command!

not what i would think of as desired behavior. I'd prefer for GPIO/Pulse commands to always be available.

workgroupengineering, 2018-06-04:
today I'm working on this issue.
But I have doubts about some commands.

| Command | From |  To|
| :---         |      ---:|          ---: |
| `gpio`    | _P001_Shitch.ino | Command.ino |
| `pulse`    | _P001_Shitch.ino | Command.ino |
| `longpulse`    | _P001_Shitch.ino | Command.ino |
| `longpusle_ms`    | _P001_Shitch.ino | Command.ino |
| `servo`    | _P001_Shitch.ino | ? |
| `status gpio`<sup>1</sup>    | _P001_Shitch.ino | Command.ino |
| `gpio pinstate`<sup>1</sup>    | _P001_Shitch.ino | Command.ino |
| `monitor gpio`<sup>2</sup>    | _P001_Shitch.ino | Command.ino? |
| `rtttl`    | _P001_Shitch.ino | ? |
| `tone`    | _P001_Shitch.ino | ? |

- `servo`, `rtttl`,  `tone`  are core commands?
- <sup>1</sup> `status gpio` and `gpio pinstate` have a similar behavior but, `gpio pinstate` `status gpio` reads directly from the I / O while `status gpio`  using the cache (getPinStateJSON).  We should find common behavior, otherwise bugs could occur.
- <sup>2</sup> `monitor gpio` The implementation of this command involves changes to the main loop, I wonder, if it is not better to use it as a plugin

TD-er, 2018-06-04:
Looking at this list, a lot of work has to be done.
Better split it into several parts (issues) and first revert the gpio behavior back to what it was. (running commands on not active plugins)

Thus:

1. Revert check for active plugins when running commands
2. Move GPIO related commands to `command.ino`
3. Make some informed decision about whether or not to use cached data or not. (maybe current implementation is fine, but should be make more clear)
4. Move other commands to `command.ino` and have a good look at them. Some are quite buggy.
5. Decide on state of "notification" like commands (tone/rtttl), if they should be a command or a notification.
6. Return check for only run commands on active plugins

I guess all these should be separate issues and maybe even be developed on a separate test branch.

Grovkillen, 2018-06-04:
We need to do this methodically, I agree with @TD-er

uzi18, 2018-06-04:
3th - add option to scan also disabled plugins ;)

TD-er, 2018-06-04:
@uzi18 That's what you get when some things are solved as an "easy quick fix". They always return as an issue.
It doesn't make any sense to have global actions inside a specific plugin.
The same with the MQTT import. I understand why it was implemented like that at the time, but it's a returning head-ache for stability.

uzi18, 2018-06-04:
We must move it to core and reimplement it as plugincall (gpio set/get request) in all related plugins, this is right way, so all problems will go into one place to fix.
I'm more offline from some time as Sloeber/Eclipse does not compile ESPEasy.
I need to investigate it but have no free time now

TD-er, 2018-06-04:
Why in related plugins? Isn't the whole idea the commands are global and thus not plugin-related?
Only problem is that setting a GPIO pin could affect operations of other plugins. That's one of the reasons why I am reluctant about "a quick fix" and have to think about this very well.

So I'm very curious about what you have in mind.

TD-er, 2018-06-05:
I merged a commit that should allow Gpio and pwm commands again like it was.
See https://github.com/letscontrolit/ESPEasy/pull/1485

workgroupengineering, 2018-06-06:
I see it #1485 now
A potential problem of the PR #1485 is if a plugin manage TaskIndex they could behavior abnormally.

I was developing a [PR](https://github.com/workgroupengineering/ESPEasy/tree/HotFix/Issue_1477) but I interrupted because I realized that ExecuteCommand is becoming too difficult to maintain.

I would like to propose something like this pseudo code for core commands:

```c++

typedef bool (*ExecuteCommandCall)(struct EventStruct *event, String& str);

class CommandsManager
{
    private:
    std::map<const char *,ExecuteCommandCall> commandRegistry;
    public:
    bool Register(const char * id, ExecuteCommandCall command);
    bool Execute(const char * id, struct EventStruct *event, String& str);
};

```

Separate the commands by competence, system (Commands / System.ino), relogole (Commands / Rules.ino), gpio (Commands / gipo.ino), etc.

#### gpio.ino
```C++
bool Command_GPIO (struct EventStruct *event, String& str)
{
   bool success = false;
   ....
   return success;
}

bool Command_PWM (struct EventStruct *event, String& str)
{
   bool success = false;
   ....
   return success;
}

....
```

#### EspEasy.ino
```C++
....

CommandsManager cmdMgr;
void Register_Commands()
{
    .....
    cmdMgr.Register(F("gpio") ,&Command_GPIO);
    cmdMgr.Register(F("pwm")  ,&Command_PWM );
    .....
}

void setup()
{
   ....
   Register_Commands();
   ....
}

```

#### Command.ino
```C++
void ExecuteCommand(byte source, const char *Line)
{
  checkRAM(F("ExecuteCommand"));
  String status = "";
  boolean success = false;
  char TmpStr1[INPUT_COMMAND_SIZE];
  TmpStr1[0] = 0;
  char cmd[INPUT_COMMAND_SIZE];   
  cmd[0] = 0;
  struct EventStruct TempEvent;
  TempEvent.Source = source; 
  GetArgv(Line, cmd, 1);
  if (GetArgv(Line, TmpStr1, 2)) TempEvent.Par1 = str2int(TmpStr1);
  if (GetArgv(Line, TmpStr1, 3)) TempEvent.Par2 = str2int(TmpStr1);
  if (GetArgv(Line, TmpStr1, 4)) TempEvent.Par3 = str2int(TmpStr1);
  if (GetArgv(Line, TmpStr1, 5)) TempEvent.Par4 = str2int(TmpStr1);
  if (GetArgv(Line, TmpStr1, 6)) TempEvent.Par5 = str2int(TmpStr1);
  
  bool success = cmdMgr.Execute(cmd, &TempEvent, line);
  yield();
  if (success)
    status += F("\nOk");
  else
    status += F("\nUnknown command!");
  SendStatus(source, status);
  yield();
}
  
```

TD-er, 2018-06-07:
Sorry, forgot to read it thoroughly.
I opened your suggestion yesterday on my phone and decided it needed some extra thoughts, so I didn't open it on github, but apparently if I open the mail on my mailclient, for long enough, it is marked as read on Github too.

And I cannot mark things as "unread" on github :(

Apart from that, there were some other real-world issues that demanded some attention and way too much energy from me.  (about Dutch politics, and our houses)

> I was developing a PR but I interrupted because I realized that ExecuteCommand is becoming too difficult to maintain.

That's always a good moment to stop and take some moment to (re)consider :)

About your CommandsManager;
I think I get it why you want to make some map from a char pointer to command call. 
My only remark is that I personally don't like such constructs, simply because it takes me more time to read it and wrap my head around it.  I'm just a simple guy ;)
Doesn't mean it isn't a resource-friendly and viable solution, it is just that it takes some of my resources to understand ;)
But if it is working well so nobody has to look at it again, then it's fine :)
Just make sure you document it so anyone can understand it, even after a year.

And I'm not sure `F("GPIO")` will be matched the same as `somestring.c_str()`, even though the compiler may accept it.
Maybe it is still the easiest to have some enum value for the map and some (memory)efficient function to match the given command to an enum.

But I guess I will need some extra look at it later this evening.

workgroupengineering, 2018-06-08:
>Maybe it is still the easiest to have some enum value for the map and some (memory)efficient function to match the given command to an enum.

for the enum you are fully right but, the problem is when adding a new command you have to update the enum then the map between string and enum with the possibility of errors.

I have not had the opportunity to test the effincency because in the little free time I have not yet completed the porting of the commands.

In this regard I have a problem regarding the SendTo and SendToUDP commands are quite similar, do I merge them ?. What do you think?

TD-er, 2018-06-08:
Hmm not sure.
Normally I would say merge duplicate code, but the UDP stuff may need a second look in the near future.
There have been numerous reports on issues with UDP lately.  (not sure if it is UDP related, or just about the inter-ESPeasy-communication)

workgroupengineering, 2018-06-19:
[Here](https://github.com/workgroupengineering/ESPEasy/tree/feature/CommandsManger), I created a branch in which I implemented a new command manager.

I tested my configuration for a few days without detecting problems.

I would like to use PSTR () to optimize memory usage, but I have a LoadStoreErrorCause exception at runtime.

Can you suggest me how to solve it?

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er