letscontrolit/ESPEasy #1478: Easyesp plugin for gsm modem

gdrako, 2018-06-03:
Hello 

Is there a future plan for adding an plugin that manages to connect a gsm modem so the module can access the internet via gprs ?

Thank you.

TD-er, 2018-06-03:
As far as I know, there are no such plans.
I guess there are so many wifi => GPRS modules available, that demand for it is a bit low. Those are perhaps even less expensive than the separate GPRS modules you could connect via serial.
Maybe SMS-like notification in areas with low 3G/4G reception or nodes that need to be low-power is something that would be more interesting.

Grovkillen, 2018-06-04:
Yes we are not focusing on a plugin for GPRS, maybe someone else find the idea interesting and are willing to spend time developing a plugin like that. But as @TD-er mention, 4G battery powered routers are available and could be used by multiple units simultaneously so I find the need for a ESP Easy plugin not so great.

But if someone else want to add it feel free! :)

sparkescreative, 2018-06-27:
I would really like a plugin for a GSM module like a SIM800L 
[image]
So that ESPEasy was able to send simple 'SendToHTTP' variables via a 'data sim card'. Adding Location data Longitude and Latitude would be incredible and very useful.

I have several projects using ESPEasy like my Lights, Heating, Security Door & Window sensors etc etc. They all log to to a server in my home and to a server online. I've got Web App's, Custom CSS for the ESP's and little pods around my house with little screens on displaying the temperature etc...

[image]

The only thing missing is an easy way to get data off through a standard phone sim in a field in the middle know where.

Any help in this direction would be greatly received.

TD-er, 2018-06-27:
Copy of my reply to the [forum](https://www.letscontrolit.com/forum/viewtopic.php?f=12&t=5505&p=29218#p29218) about this:

Adding another communications device will have quite some impact on a large number of files.
For example you have to detect whether or not there is an active connection and how to make and manage such a connection.
Also you probably want to limit the data stream when on a metered connection like a GSM module.

The easiest solution is probably adding a new controller which will manage the connection and data limits.
But that makes it rather inflexible when you want to upload to let's say Domoticz, OpenHAB and the platform you're using.

So there needs to be a new layer to manage the connection and both the webserver and the controllers need to be able to select what communication module should be used.

In short:
It would be rather easy to add minimal support for such a module for your specific use case.
But such a solution would be a big pain to maintain and support.

A proper solution would take quite some development effort.

garudaonekh, 2021-09-04:
Sorry to dig up a 3 year old post.
I use TTGO T-Call(ESP32+Sim800L) for managing sensors and relay at farm.
Blynk has very close link with TinyGSM(i guess because Blynk cofounder is also TinyGSM founder) as as far I know it makes GSM to Blynk a very easy integration.
`
  Blynk.begin(auth, modem, apn, user, pass);`

I test run it several days, i see it work very smooth thru my 2G SIM.

So my question is can we just add this to Blynk Controller(C015) in ESPEasy so that I can setup ESPeasy on my T-Call and use blynk cloud for my mobile app.

Thansk;