letscontrolit/ESPEasy #1479: No access to Webui after disconnect to a wifi network

wolverinevn, 2018-06-04:
### Summarize of the problem/feature request

Flash ESPEasy Mega, connect to Espeasy_0 to config wifi ssid and wpa2-key (no fallback), successfully connected. Turn off wifi ssid, reboot EspEasy, connect to Espeasy_0 again. Web ui appear for the first access, move to Config tab -> no response at all.
Reboot ESPEasy -> only first access again.
### Expected behavior

ESPEasy webserver should let me go to config tab to config another wifi ssid.

### Actual behavior

Webui does not response after the first access
### Steps to reproduce

1. Disable your wifi
2. Connect to ESPEasy's self-wifi
3. Access 192.168.4.1

Yes

### System configuration

Hardware:
NodeMCU 12E

ESP Easy version: mega-20180524, mega-20180522, mega-20180306

ESP Easy settings/screenshots: 
No specific settings so far, except ssid and ap password.
### Rules or log data

```

```

wolverinevn, 2018-06-04:
@TD-er Yep, I disabled wifi station then reboot ESPEasy.

mega20150522 solved the issue. I still have no idea why it happened and solved. I have a few NodeMCU 12E(s) running old ESPEasy mega firmware and have never seen this issue.

Grovkillen, 2018-06-04:
So this is how you normally do the configuration? I guess it should be possible to access the GUI during AP mode but you should connect to the unit by just letting it reboot with the new SSID/password. You seem to do it too complicated.

PR #2559 (2019-08-26): [WiFi] Improve WiFi reconnect stability and fix initial setup
Simplified the WiFi connect state machine.
Fixed setup of initial WiFi configuration.

AP mode is now kept as long as client is connected.
AP mode will be started immediately if there are no valid WiFi credentials, or after 20 seconds of no wifi connection.
AP mode will be disabled 60 seconds after starting it, or after the last client disconnected from the AP.

Fixes #2561 
Fixes #2338 
Fixes #2203 
Fixes #1987 
Fixes #1737
Fixes #1640 
Fixes #1565
Fixes #1479 
Fixes #1098
Fixes #558
closed 2019-08-26 by TD-er