letscontrolit/ESPEasy #1480: Inter-ESPEasy Network offline ?

s0170071, 2018-06-04:
The two units that I flashed two weeks ago are not using the inter-espeasy network any more. The older units do, the two "new" ones don't. They do not even see themselves...
Is this a known issue ? I thought I read something somewhere but can't find it.

TD-er, 2018-06-04:
Nope, is not a known issue.

Are you sure you've got the UDP port set and all have an unique unit-id?
Last night I did flash an ESP32 node with the latest source and that one could see others. (just tested to make sure and it still works)

You also have to reboot (reset or cold boot) to make it work properly.

s0170071, 2018-06-04:
Got it. It appears I am a victim of UDP packet loss. Both units sit in a very poorly covered area of my house. After a reset they show faint signs of inter-ESP network activity. 

Btw, is there any use for setting a ESPEasy ID ? Why don't we use the last byte of the IP address for that?
Or - consider that last byte internally so you can actually have identical IDs ? That would eliminate that source of error forever.

TD-er, 2018-06-04:
We have to look into this global "know eachother" option.
Currently there is an array of structs of 32 elements to keep track of other nodes.
If my memory serves me well, these are used to perform faster 'global sync' like communications. Those are recently moved to Controller c013.

closed 2023-09-04 by TD-er