letscontrolit/ESPEasy #1482: feature request: sunrise/sunset in rules

TungstenE2, 2018-06-05:
### Summarize of the problem/feature request

I would like to use sunrise/sunset time in a rule (eg sunrise + 30min) on a Wemos D1 mini.
How to do this? An external HTTP request would be good, but not to a lokal controler.
I want to be independent from another controler.

In Tasmota this feature was implemeted as well: https://github.com/arendst/Sonoff-Tasmota/issues/2461

Any ideas or tipps?

thx

### Expected behavior

Similar to tasmota solution or a setting for the location and a http call to fetch sunrise and sunset from the internet.
There are services that provide such information.
e.g: https://www.timeanddate.com/astronomy/germany

### Actual behavior

n.a.

TD-er, 2018-06-05:
Creating an event to be used in rules is quite simple.
Then it is easy to do something at event + X minutes. But if something X minutes before event is needed, then it should be set as a special time in the rules itself. 
For example similar to "Sun 23h", but then something like "sunrise-30m" or "sunset+30m".
I guess that's the most flexible way to implement it.

TungstenE2, 2018-06-05:
but where to get sunrise/sunset time from? it changes every day and for every location it is different.

TD-er, 2018-06-05:
Just let the user add latitude/longitude coordinates and then we can compute it.
The date is known from NTP.

rieders, 2018-06-08:
Hello

I would also be interested in such a plugin.
I have an aquarium and I would run the esp with WS2812 LEDs.
Then there would have to be considered the number of LEDs.
As well as an offset for sunrise and sunset times.

I would be happy if such a thing can be realized. I am looking for a long time for a lighting solution for my saltwater aquarium.

Greetings André

TungstenE2, 2018-06-21:
In Release 20180618 GPS coordinates were added to Advanced Settings. Great!

What else needs to be done in order to work with this and to get sunrise/sunset working for rules?

https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20180618

TD-er, 2018-06-21:
@TungstenE2 You can already use the exact times in rules.
That makes it possible to do something at sunrise or sunset and also start a timer to do something at those moments + N seconds.
However, you also would want to do something at N minutes before those events and setting it directly to  sunrise+N minutes would also be nice.

So I would like to add generic variables like `%sunrise+5%` and `%sunset-15%`.  (numbers in minutes)
This means an extension to the variables parser, which is not there yet.

TungstenE2, 2018-07-04:
I was just about to test it.

New %sunrise% and %sunset% variables were added, but I am not sure about the syntax to be used in rules.

Is there a way to test the variables?

https://www.letscontrolit.com/wiki/index.php/ESPEasy_System_Variables

TD-er, 2018-07-04:
The easiest way to see their values is to output them to a display or in a topic via MQTT.

Maybe we should add some help page in the web interface showing all variables and their current values.

TungstenE2, 2018-07-04:
that would be a great help!

TungstenE2, 2018-07-04:
created feature request: [issue 1550](https://github.com/letscontrolit/ESPEasy/issues/1550)

TungstenE2, 2018-07-04:
could you give an example rule how to use %sunrise% and %sunset% variables?

TD-er, 2018-07-06:
@TungstenE2 I added it to the [wiki](https://www.letscontrolit.com/wiki/index.php/Tutorial_Rules#.28System.29_events)

TungstenE2, 2018-07-06:
just testing with timers

```
//at sunrise time set timer1 for 1 hour
On Clock#Time=All,%sunrise% do 
        timerSet,1,3600                               
endon

//When Timer1 expires, <action>
 On Rules#Timer=1 do  
    <action>
 endon
 
 
 //at sunset set timer1 for 0,5 hour
On Clock#Time=All,%sunset% do 
        timerSet,2,1800                               
endon

//When Timer2 expires,  <action>
 On Rules#Timer=2 do  
    <action>
 endon
```

TD-er, 2018-07-06:
I will add extra variable options like `%sunrise-30%` which will return a timestamp 30 minutes before sunrise.
Those negative offsets cannot be done using timers. 
Or you should set a timer to something like (24h - 30m), but that may lead to other issues like almost a day latency and up-to -3 ... +3 minutes offset over the year.

susisstrolch, 2018-07-07:
How can I see the actual values for %sunset% (f.e. In log or browser)?

Grovkillen, 2018-07-07:
Not yet, in time.

TropicMike, 2018-07-14:
I would love to see %sunset% - 2h or something like that.  Sunset is technically too dark here in my house and I'd like to turn my living room light on maybe 1 or 2 hours prior to sunset.  If I missed something and this is possible already, please let me know - this is my first ESPeasy project (using Sonoff Basic for an auto-setting sunset lamp timer).

M0ebiu5, 2018-07-15:
@TropicMike one possibility would be to set a 22h timer on sunset (i see now that TD-er already described this 3 posts before - the few minutes deviation are negligible, cause coordinates and calculation are also not precise)

TD-er, 2018-07-15:
@M0ebiu5 There is one other issue with that approach... It is needed to have no reboot of the ESP during those 22h. Or else the rule will not work.

M0ebiu5, 2018-07-15:
@TD-er Thats right! Are you afraid that espeasy will suffer from an unintentional reboot during that 22h? :)
Just joking - i see you are trying hard to improve stability...

Cause he wants to use it to just turn on a light - a fallback rule with a hard-coded time would help. Although for a proper time on a sonoff, it needs network and ntp, so there are many more possibilities to turn it on at the right time...

TD-er, 2018-07-15:
@M0ebiu5 Yep, stability is sometimes an issue.
Stability on the electricity grid, of course :)

But this compute with minutes offset will get there.

TropicMike, 2018-07-15:
Thanks for the tips, everyone!  I don't need minute-accuracy by any means - I'm just looking to have the lamp adjust as the weeks go-on so my wife isn't complaining that the mechanical timer is "too late" or "too early" like she does now ;)

TD-er, 2018-07-19:
It is now implemented.
See: https://github.com/letscontrolit/ESPEasy/pull/1587

closed 2018-07-19 by TD-er

uzi18, 2018-07-19:
You can use also this now, without timer:

//at sunrise time
On Clock#Time=All,%sunrise% do
     <action>

endon

//, <action>
On Clock#Time=All,%sunrise+30m% do

    <action>
endon

TD-er, 2018-07-19:
Yep, that was the whole idea. It now also works right after a reboot, as long as NTP is working in time :)

TropicMike, 2018-07-19:
Wow - thank you!  I'll have to grab the latest version and try it out!

I'm a GitHub noob - is there a way to see which version this change is committed into?  I'd like to learn how to determine that for myself so I don't have to ask in the future.

TD-er, 2018-07-19:
This is "core" stuff, so it is included in all build-files.
The dev/testing etc is more about included plugins and controllers.
Sometimes a plugin could crash the entire system, even when not enabled.

Also the dev/testing versions are using more memory and the extra plugins are not very well tested.

TropicMike, 2018-07-19:
Thank you for the clear explanation, TD-er!

jodur, 2018-11-23:
If this is "core"stuf why isn't it released in the stable V2.0xx version?

TD-er, 2018-11-23:
We are working towards a new stable release.

Grovkillen, 2018-11-23:
@jodur there is no stable release (yet). the "2.0" branch is abandoned, but we are not far away from releasing the proper stable 2.0. We might go ahead and call it 2.0.1 or 2.1 or something.

jodur, 2018-11-23:
Wow quick response. Thx for all the good both have done at first!
The info about release on the [Wiki](https://www.letscontrolit.com/wiki/index.php/ESPEasy) is a little bit confusing (Q3). Any ETA for the stable release?

Grovkillen, 2018-11-23:
The wiki is really outdated and I haven't put any love into it since we're currently transforming it into a read the docs syntax. It will be much easier to maintain. You can start getting familiar with it here: https://espeasy.readthedocs.io/en/latest/

It's still a major work in progress but each day I add more stuff and when that is 100% transferred we hope that we can release a stable (we are really close to have a stable firmware but need to have the documentation up to par with it before we can press the release button).