letscontrolit/ESPEasy #1484: Multiple Ultrasonic sensors = same reading on both.

waspie, 2018-06-05:
### Summarize of the problem/feature request
If you make 2 Ultrasonic devices both will have the same reading

### Expected behavior
they shouldn't have the same reading

if added separately each will read correctly

### Actual behavior
they have identical readings

### Steps to reproduce
1. add two ultrasonic sensors
2. not sending to controller. 
3. set on different times or the same, doesn't matter

yes

yes, no change. 

### System configuration
nodemcu v3 

ESP Easy version: may 24 2018 build

ESP Easy settings/screenshots: 
[image]
[image]

waspie, 2018-06-05:
I rolled by own using the SR04 plugin from April 14 2018 (had to make two small changes) and it works properly. Something with the Newping/updated plugin broke it :\

TD-er, 2018-06-05:
@waspie What has to change?

waspie, 2018-06-05:
A couple lines in the webform section. to be clear, i made the edits in the old, non-newping sr04 plugin.
assuming you want to continue on with the newping one, right?

waspie, 2018-06-05:
OLD PLUGIN line 66 replace with:
addFormSelector(F("Mode"), F("plugin_013_mode"), 2, options, optionValues, choice);

OLD PLUGIN line 66 replace with:
addFormNumericBox(F("Threshold"), F("plugin_013_threshold"), Settings.TaskDevicePluginConfig[event->TaskIndex][1]);

then it compiles fine on MAY 24th 2018 source and works fine with multiple US sensors.

TD-er, 2018-06-05:
@waspie Those lines are exactly the same as the current implementation.

Maybe you forgot to add the new lines? ;)

TD-er, 2018-06-05:
I would expect there needs to be more than one sonar object, one for each active plugin using this NewPing

waspie, 2018-06-05:
Those lines were transplanted into the plug-in from April 14th prior to the
move to beeping. I sourced them from the plug-in in the current release.

I'm using the old plug-in with the current source

TD-er, 2018-06-05:
I just wrote something, which may work. Could you test it?
I don't have these sensors here, or at least not connected to my ESP nodes (need resistors to manage line levels)

It is not tested, just a simple test to see if it would compile.
I guess there is still some issue when deleting one sensor, that it the other one will not work until it is saved again (or reboot).
So just a first test to see if this is the correct approach.
If it is, I will make some generic template function to be used in all plugins that may need multiple instances.
A lot of plugins have issues when using >1 of them.

```C++
#ifdef USES_P013
//#######################################################################################################
//############################### Plugin 013: HC-SR04, RCW-0001, etc. ###################################
//#######################################################################################################

#define PLUGIN_013
#define PLUGIN_ID_013        13
#define PLUGIN_NAME_013       "Distance - HC-SR04, RCW-0001, etc."
#define PLUGIN_VALUENAME1_013 "Distance"

#include <Arduino.h>
#include <vector>
#include <NewPing.h>

struct P_013_sensordef {
  P_013_sensordef(byte TRIG_Pin, byte IRQ_Pin) : P_013_TRIG_Pin(TRIG_Pin) {
    sonar = new NewPing(TRIG_Pin, IRQ_Pin);
  }

  ~P_013_sensordef() {
    delete sonar;
    sonar=NULL;
  }

  byte P_013_TRIG_Pin;
  NewPing *sonar;
};

std::vector<P_013_sensordef> P_013_sensordefs;

boolean Plugin_013(byte function, struct EventStruct *event, String& string)
{
  static byte switchstate[TASKS_MAX];
  boolean success = false;

  switch (function)
  {
    case PLUGIN_DEVICE_ADD:
      {
        Device[++deviceCount].Number = PLUGIN_ID_013;
        Device[deviceCount].Type = DEVICE_TYPE_DUAL;
        Device[deviceCount].VType = SENSOR_TYPE_SINGLE;
        Device[deviceCount].Ports = 0;
        Device[deviceCount].PullUpOption = false;
        Device[deviceCount].InverseLogicOption = false;
        Device[deviceCount].FormulaOption = true;
        Device[deviceCount].ValueCount = 1;
        Device[deviceCount].SendDataOption = true;
        Device[deviceCount].TimerOption = true;
        Device[deviceCount].GlobalSyncOption = true;
        break;
      }

    case PLUGIN_GET_DEVICENAME:
      {
        string = F(PLUGIN_NAME_013);
        break;
      }

    case PLUGIN_GET_DEVICEVALUENAMES:
[... 96 lines cut ...]
              sendData(event);
            }
          }
        }
        success = true;
        break;
      }
  }
  return success;
}

/*********************************************************************/
float Plugin_013_read(byte TRIG_Pin)
/*********************************************************************/
{
  for (unsigned int i = 0; i < P_013_sensordefs.size(); ++i) {
    if (P_013_sensordefs[i].P_013_TRIG_Pin == TRIG_Pin) {
      return P_013_sensordefs[i].sonar->ping_cm();
    }
  }
  return 0;
}
#endif // USES_P013

```

waspie, 2018-06-06:
i will test this in the next few hours, thank you

waspie, 2018-06-06:
ohh boy she is not happy! 
`04 : WIFI : Connecting ha_chicken attempt #0
208 : WD   : Uptime 0 ConnectFailures 0 FreeMem 27728

Soft WDT reset

ctx: cont 
sp: 3fff2550 end: 3fff29e0 offset: 01b0

>>>stack>>>
3fff2700:  3ffe8cdc 00000000 3fff4124 40231895  
3fff2710:  3fff5364 0000000f 3fff4124 402318ff  
3fff2720:  3fff18cc 0000048e 3fff2870 40231948  
3fff2730:  00000002 3fff2870 3fff2760 4020e4c1  
3fff2740:  3fff18cc 0000048b 0000048b 402245aa  
3fff2750:  00000002 3fff2870 3fff2780 40100690  
3fff2760:  3fff18cc 00000488 00000488 4010020c  
3fff2770:  00000002 3fff2870 3fff27a0 40100690  
3fff2780:  00000000 00000000 00000000 40232f30  
3fff2790:  3ffe97c8 00000000 4026a1c8 4020a726  
3fff27a0:  00000000 00000000 00000000 3fff0af4  
3fff27b0:  3ffefe50 3fff2870 00000003 4020ee9e  
3fff27c0:  00000000 3fff0000 00000000 3fff1801  
3fff27d0:  00000000 00000000 00000000 00000000  
3fff27e0:  00000000 00000000 3fff4004 0000000f  
3fff27f0:  00000000 3fff401c 0000000f 00000000  
3fff2800:  3fff5304 0000000f 00000000 3fff531c  
3fff2810:  0000000f 00000000 3fff5334 0000000f  
3fff2820:  00000000 00000000 3fff28c8 4023300d  
3fff2830:  3ffefda4 00000000 3fff28c8 4023305a  
3fff2840:  3ffe97c8 00000000 3fff2870 40100690  
3fff2850:  00000000 00000000 3fff2870 3fff0af4  
3fff2860:  3fff0564 3ffeffd8 3fff0e94 402231b2  
3fff2870:  00000000 00000000 00000000 3fff1801  
3fff2880:  00000000 00000000 00000000 00000000  
3fff2890:  00000000 00000000 3fff4094 0000000f  
3fff28a0:  00000000 3fff2fcc 0000000f 00000000  
3fff28b0:  3fff460c 0000000f 00000000 3fff4964  
3fff28c0:  0000000f 00000000 3fff48f4 0000000f  
3fff28d0:  00000000 00000000 42000000 00000000  
3fff28e0:  00000000 00000000 3ffefc28 3fff19b4  
3fff28f0:  3fff18cc 00000018 00000018 4010020c  
3fff2900:  3fffdad0 00000000 3fff2930 40100690  
3fff2910:  3ffe9738 00000000 402741b0 40232f30  
3fff2920:  4020a27a 00000000 000075d9 4020b34b  
3fff2930:  00000000 00000000 00007530 00000001  
3fff2940:  3fff0ea4 00000000 3ffeff48 402232de  
3fff2950:  00000000 0000047f 000004b5 4020b34b  
3fff2960:  3fffdad0 00000000 000003e8 3fff19b4  
3fff2970:  3fffdad0 00000000 3ffefd34 4022880f  
3fff2980:  3fffdad0 00000000 000000cd 4020b34b  
3fff2990:  00000000 00000000 3ffefd8c 3fff19b4  
3fff29a0:  3fffdad0 00000000 3ffefd8c 40228dbd  
3fff29b0:  feefeffe feefeffe feefeffe 3fff19b4  
3fff29c0:  3fffdad0 00000000 3fff19ae 40233e98  
3fff29d0:  feefeffe feefeffe 3fff19c0 40100710  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
ªU71 : 

INIT : Booting version: (custom) (ESP82`

Grovkillen, 2018-06-06:
You did a "double-compile"? I have often problems with the first compile.

waspie, 2018-06-06:
no, i've never had an issue previously... I will take another run at it in a bit and report back.

thanks for looking into this guys

TD-er, 2018-06-06:
Any luck already?

waspie, 2018-06-06:
Give me about 2 hours. By then you'll be asleep ha

TD-er, 2018-06-06:
I'll give you enough hours to test. I'm off to bed in less than an hour from now. :)

waspie, 2018-06-07:
double tripe compiled. still crashes. you don't need to have them to crash it, just build and flash and add 2 US. on the submit of the second US it crashes

waspie, 2018-06-07:
I don't remember seeing that, my bad. in the meantime see the 5th post for a couple edits you can do to the old plugin to make it work with the current source

hiten12, 2018-06-08:
I am using the old plugin with some changes , and it works fine.

waspie, 2018-06-09:
> I added a PR.
> Could you test the PR?
>
> There is something fishy with the test for micros() in that library.
> And it is rather strange the setting for using 1 pin or 2 must be made at
> compile time.

In a few hours

waspie, 2018-06-11:
ugh, still havent tested, sorry. busy!

waspie, 2018-06-25:
I finally tested this and neither work. not 1. not 1 or 2. no good at all

TD-er, 2018-10-02:
This should be fixed now.
So I will close it.

closed 2018-10-02 by TD-er

waspie, 2019-01-02:
this still isn't fixed. i still have to take the ~april sr04 plugin and modify to get more than one ultrasonic sensor to read.

discostew2, 2022-07-12:
Hey everyone! Thank you so much for working on the _P013_HCSR04 plugin feature to support multiple sensors. I was wondering if anyone has successfully tried setting up two devices with _P013_HCSR04.ino and gotten this to work? I recently tried setting up 2x sensors (with 2x different sets of GPIO) - when either of the two sensors is attached (but not both) I receive good / valid data, however when I try to attach both sensors at the same time both devices show '0' as the result (with and error in the logs). If I unplug either of the two sensors the other starts working again and gives good data.

At first glance, it appears that the _P013_HCSR04 is doing the right thing with multiple instances, so perhaps there is something int the NewPingESP8266 library that does not support multiple concurrent instances? Has anyone attempted to test / reproduce this?

Thanks again for all of your work on this!

tonhuisman, 2022-07-12:
AFAICS, this plugin hasn't been adapted for multi-instance use, so that explains why you don't get the expected reading when having multiple sensors connected and active.

Haven't looked at the effort that's needed to fix that, but I'll put it on my TODO list 😃

TD-er, 2022-07-12:
The plugin does seem to have some std::map of NewPing8266 objects, so it looks like there has been some effort to allow for multiple instances of the same plugin.
Still I guess it wouldn't hurt to have a good look at the code.
Perhaps the NewPing8266 class does have some internal state which isn't shared well among multiple instances.