letscontrolit/ESPEasy #1488: MQTT user cannot be edited

s0170071, 2018-06-08:
using latest core and ESPEasy the MQTT user defaults to the SSID. It cannot be edited. 
One thing I did change though was incrementing DEVICES_MAX by 3 so I can use my own plugins

in ESPEasy globals.h: 

#if defined(PLUGIN_BUILD_TESTING) || defined(PLUGIN_BUILD_DEV)
  #define DEVICES_MAX                      75+3
#else
  #define DEVICES_MAX                      50+3
#endif

[image]

s0170071, 2018-06-08:
This one is nice: 
If I 
- modify the controller user, it is saved ok. 
- delete the controller user (empty field) its reset to be the SSID
- the SSID setting remains unchanged
- modify the user, I can empty the password field and it remains empty
- leave the user name unchanged (like SSID name), I cannot delete the password.

TD-er, 2018-06-08:
Just as general note, you have to be careful when increasing max. values.
Some of the structs are stored right after eachother and increasing maximums may result in overlap of settings when stored.

The `DEVICES_MAX` does have a direct relation to the amount of free memory, but should not affect other functionality.

I guess the web page does look into some values (not all) and base decisions on that.
You could try the inspect window of your browser (Network tab) and see what is returned when you press Submit.
Perhaps some of the parameters are not returned or still contain some value when meant to be empty. (something like a space)
Probably the logic in WebServer.ino is buggy on that screen.

s0170071, 2018-06-08:
When I push submit the params tab shows the correct value. When I emtpy the field, the "controlleruser" parameter is also empty.

s0170071, 2018-06-08:
Just tried another ESP, here everything is fine. Seems like a defective board.

closed 2018-06-08 by s0170071

TD-er, 2018-06-08:
You could also try to clear the entire flash.
Sometimes some left-over settings may cause issues.