letscontrolit/ESPEasy #1493: SendToHttp

rene9900, 2018-06-10:
i have a strange problem...
deep sleep reset esp8266-01 must turn on a led on another esp8266 unit on my network.

on the deepsleep unit rules:

on System#Boot do
SendToHttp 192.168.1.2/control?cmd=GPIO,12,1
endon

dont work..
but in my brpwser typing:
http://192.168.1.2/control?cmd=GPIO,12,0
does work..

what am i doing wrong?

the reciving unit info:

System Info |  
-- | --
Unit | 34
Local Time | 2018-06-10 22:19:49
Uptime | 0 days 7 hours 9 minutes
Load | 7% (LC=14372)
Free Mem | 16336 (4336 - sendWebPageChunkedData)
Boot | Manual reboot (9)
Network
Wifi | 802.11N (RSSI -79 dB)
IP / subnet | 192.168.1.2 / 255.255.255.0
GW | 192.168.1.1
Client IP | 192.168.1.50
Allowed IP Range | All Allowed
STA MAC | A0:20:A6:0D:D9:48
AP MAC | A2:20:A6:0D:D9:48
Firmware
Build | 20100 - Mega (core 2_4_0)
GIT version | (custom)
Plugins | 68 [Testing]
Md5 | 0000000000000000
ESP board
ESP Chip ID | 907592 (0xDD948)
ESP Chip Freq: | 80 MHz
Storage
Flash Chip ID | Vendor: 0xE0 Device: 0x4016
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash IDE speed: | 40 MHz
Flash IDE mode: | DIO
Flash Writes | 9 daily / 9 boot
Sketch Size | 639 kB (2432 kB free)

TD-er, 2018-06-10:
You're trying to send it to "yourself"?
At boot, there is no guarantee the network is already "up". For that you need to react on another event.
See [Event reference](https://www.letscontrolit.com/wiki/index.php/Tutorial_Rules#.28System.29_events)
So why not just run the command? Sending it over http is a bit overkill from the rules to yourself.
Also it is a bit hard to communicate in a multi-threaded behavior with just a single threaded core.

rene9900, 2018-06-11:
yes i have a magnet switch in my mailbox resetting the esp when mailman opens the mailbox, then it send to another esp in my livingroom lighting up a led on pin12

it stopped working for a unknown reason with event:
On Music#1 do
gpio 12,1
//rtttl,14:d=4,o=5,b=180:8f,8f,8f,2a#.,2f.6,8d#6,8d6,8c6,2a#.6,f.6,8d#6,8d6,8c6,2a#.6,f.6,8d#6,8d6,8d#6,2c6,p,8f,8f,8f,2a#.,2f.6,8d#6,8d6,8c6,2a#.6,f.6,8d#6,8d6,8c6,2a#.6,f.6,8d#6,8d6,8d#6,2c6
endon

TD-er, 2018-06-11:
What version of ESP easy?
20180524 had issues with some commands.

rene9900, 2018-06-11:
the reciving esp is 
Firmware Build | 20100 - Mega (core 2_4_0)
the sender esp is
Build: | 120
Core Version: | 2_4_0_RC1

nonflammable, 2018-06-14:
@rene9900
there is syntax error:
on System#Boot do
SendToHttp 192.168.1.2/control?cmd=GPIO,12,1
endon

you should use:
SendToHttp 192.168.1.2,80,/control?cmd=GPIO%2C12%2C1

and 
WiFi#Connected is better than System#Boot
but I use Time#Initialized and NTP enabled

rene9900, 2018-06-15:
@nonflammable thanks man, you are a lifesaver :) the syntax wwas indeed wrong and now the led lights up on system#boot, but i can't make it work on WIFI#Connected for some reason..
if i use syntax like this it works fine:

on System#Boot do
GPIO,1,1  //save batt power by turning onboard led off
SendToHttp 192.168.1.2,80,/control?cmd=GPIO%2C12%2C1
endon

nonflammable, 2018-06-15:
indeed i am lifesaver ... but everyday as firefighter :)

TD-er, 2018-06-15:
@nonflammable During the day he fights fire, during the night he fights bugs and is on war against syntax terror :)

rene9900, 2018-09-25:
is it possible to somehow send the cmd SendToHttp 192.168.1.2,80,/control?cmd=GPIO%2C12%2C1 via node-red?

nonflammable, 2018-09-25:
i think you should use http request:
http://192.168.1.2/control?cmd=GPIO%2C12%2C1