letscontrolit/ESPEasy #1495: BME280 does not write any information in the logfile

micropet, 2018-06-11:
Is it so desired that the BME280 does not write anything in the logfile?
All other sensors write their EVENT there.

TD-er, 2018-06-11:
With the latest update? (my commit from yesterday?)

I ran it for about 8 hours before committing and it ran fine.
No event is written in the log? or fewer?
Can you see if it is sending every update (to Domoticz or whatever domotica tool you use) or significant less than desired?

It should log every sample to the log.

micropet, 2018-06-11:
Hi,

yes with the latest and all older versions.
I mean only the BME280 Events.
All other Events are fine.

micropet, 2018-06-11:
No, that has always been that way. You did not break anything.

TD-er, 2018-06-12:
I just checked and the BME280 plugin does write to the log file as `LOG_LEVEL_INFO`.
However, when using an BMP280, the humidity value may have been un-initialized.
I just fixed that.

I also changed the output to Domoticz when a BMP280 is connected.

micropet, 2018-06-13:
I have WEB LOG on Info and get no log entries from the BME280
But the BH1750 writes to the Log.

Grovkillen, 2019-10-10:
This is fixed right? If not open it.

closed 2019-10-10 by Grovkillen