letscontrolit/ESPEasy #1496: Problem with two step OTA Sonoff S20

melwinek, 2018-06-11:
I flash with OTA: ESPEasyUploaderMega, next i flash with him ESP_Easy_mega-20180611_normal_ESP8266_1024.bin
The result is no wifi connection and no AP.
I had to flash by serial.
Maybe the bin file in current releases is too big?
In ESPEasyUploaderMega is info: Sketch Max Size 604kB, but current release has 682kB.

TD-er, 2018-06-11:
The 'normal' build has been larger than 604 kB for a while already.
The "hard_SONOFF_POW" build is about 560 kB and that one has just about anything you can have on a Sonoff, including all controllers.  (all without soldering)

My initial idea was to create some builds special for devices like Sonoff's, which use only the code for the devices on board and preferably some presets to configure all to work almost out-of-the box.
Maybe we should make some make build special for Sonoff but then with some preset selection to make sure we don't have to build 10 Sonoff builds.

At least the current hard_SONOFF_POW build proves it is very well possible to stay below this 604 kB size limit.

melwinek, 2018-06-12:
Is it possible that the MEGAUploader does not have the AP WIFI function?
If there really is not, maybe it would be possible to add?
Because sometimes when I change from my own compilation to the original release, I need to reset the wifi settings, otherwise it does not work. Then with a two-stage OTA, AP WIFI in MEGAUploader would be useful.

uzi18, 2020-04-06:
@TD-er there is a hope - in arduino core development branch gziped image support was added to eboot, but we will have to provide such bins in release.

TD-er, 2020-04-07:
Yep and it promises to give roughly 70k gain in max. storage.
We can also take a look at the config.dat file for 1M nodes as we can strip off the trailing 0's and get it down to 36k.
So this means we may even reduce the SPIFFS to 64k (or 96k if there is a definition for that)
The 2-way OTA should then have support for GZip and we must remove SPIFFS support in the 2-step OTA image (or else it will not fit)
We can store the WiFi credentials in the EEPROM section right before the SPIFFS which is now not yet used.