letscontrolit/ESPEasy #1497: Crash loop when enabling NTP

stephjj, 2018-06-11:
ESPEasy version : 20180606
Hardware : Wemos D1 mini

After enabling NTP (fr.pool.ntp.org), the device crash and loop

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
   ▒U87 :

INIT : Booting version: mega-20180606 (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
88 : INIT : Warm boot #13
89 : FS   : Mounting...
114 : FS   : Mount successful, used 75802 bytes of 957314
416 : CRC  : program checksum       ...OK
446 : CRC  : SecuritySettings CRC   ...OK
552 : INIT : Free RAM:24496
552 : INIT : I2C
552 : INIT : SPI not enabled
1380 : INFO : Plugins: 47 [Normal] (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
1380 : WIFI : Set WiFi to STA
1412 : WIFI : Connecting Masaq attempt #0
1433 : WD   : Uptime 0 ConnectFailures 0 FreeMem 23784

Exception (28):
epc1=0x402083c9 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: cont
          sp: 3fff33e0 end: 3fff3660 offset: 01a0

>>>stack>>>
           3fff3580:  00000002 0000002f 3fff15d0 4024c407
3fff3590:  3ffe8f27 00000000 3fff15d0 00000027
3fff35a0:  3fff56aa 3fff56ab 3fff14e0 4024c484
3fff35b0:  3ffe0034 3fff2574 3fff2574 4024b654
3fff35c0:  00000002 3fff567c 3fff14e0 40213789
3fff35d0:  0000002f 00000000 3fff0be8 3fff2634
3fff35e0:  3fffdad0 00000000 3fff3610 3fff2634
3fff35f0:  3fffdad0 00000000 3fff0be8 3fff2634
3fff3600:  3fffdad0 00000000 3fff0be8 4023613e
3fff3610:  3fff567c 0000003f 0000002f 4020bdcc
3fff3620:  00000000 00000000 3fff0a5c 40238535
3fff3630:  feefeffe feefeffe feefeffe 3fff2634
3fff3640:  3fffdad0 00000000 3fff262e 4024cfd4
3fff3650:  feefeffe feefeffe 3fff2640 40100700
<<<stack<<<

TD-er, 2018-06-11:
Is it a build you made yourself, or one we made? If so, what filename did you use? 

I've seen such crashes before, but often they were a bad build and it was fixed when building twice.
Does it also happen when setting no NTP-server hostname?  (then it will pick random from the pool)

stephjj, 2018-06-11:
It's the official release ESPEasy_mega-20180606
I use ESP_Easy_mega-20180606_normal_ESP8266_4096.bin to flash the Wemos.

You said that I can enable NTP without a NTP Hostname ? I will try when I manage to regain control of the hardware. Reflashing with the same binary is not enough.

TD-er, 2018-06-11:
esptool can perform a complete wipe of all flash.
This means you loose all settings then.

But I guess it is not just the NTP setting, because it is not even connected to WiFi, according to the log.

stephjj, 2018-06-12:
I will try to wipe everything because it's an upgrade from ESPEasy_v2.0.0-dev12. May be there is some older conf in the flash

waspie, 2018-06-12:
IIRC coming from a version of that age it is a necessity to erase!

stephjj, 2018-06-13:
I have bricked my Wemos trying to erase flash.
I think I cannot reproduce the problem.
I will buy another one....

TD-er, 2018-06-13:
It is quite hard to really break the flash.
Maybe you could connect GPIO-0 to GND, like the flash button on NodeMCU modules.
Don't forget to remove this connection after flashing and perform a reset. (power cycle or press reset button)

stephjj, 2018-06-17:
It's what I have done, but the flashing is very very quick and the ESP doesn't start anymore.
Only the first line of the bootrom : ets Jan 8 2013,rst cause:2, boot mode:(3,6) and nothing else.

Grovkillen, 2018-06-24:
I don't think it's bricked. I've had that same serial output many times when flashing units. Try some more, try with reset button pressed (release after a couple of seconds after flash is initiated).

TD-er, 2018-06-24:
@Grovkillen You mean the flash button?
The Wemos D1 doesn't have a flash button, only a reset.

Grovkillen, 2018-06-24:
Sorry, meant the reset button.

stephjj, 2018-07-14:
The solution to stop the crash loop is to format the SPIFF (ie remove the config files)

The source of the crash is not NTP but SSDP
I close this issue and reopen with a correct title.

closed 2018-07-14 by stephjj