letscontrolit/ESPEasy #1502: Deepsleep | reconnecting to wifi - issues

m-anish, 2018-06-13:
Deepsleep reconnection to wifi waits only for 6 seconds. This causes intermittent problems. See the following links. 

1. https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=5462
2. https://github.com/letscontrolit/ESPEasy/commit/7d61e55d3442c50fdc105b6bc7b3bd638169eed2

Changing the timer value in Misc.ino from 6000 to 12000 fixes ALL reconnection issues at my end. 

I don't think this is a proper fix, but it is a workaround for now.

TD-er, 2018-06-13:
You can set the "awake time" in the advanced settings. If you set that to 12 seconds, is it then working like it should?

m-anish, 2018-06-14:
No, it does not. Even if awake time is set to 20, the module will go to sleep in 6-7 seconds if it can't connect to wifi.

TD-er, 2018-06-14:
And with the changed code (included in last night's build) ?

TD-er, 2018-06-14:
@m-anish I don't understand what you mean with the last post.
Did I miss something?
Or do you mean changing something in the commit dating back March 31 ?

m-anish, 2018-06-15:
I meant a small change in the code/commit.

https://github.com/letscontrolit/ESPEasy/commit/f0aa9c50f6f9986fd412551e85619c5fc0885bad

On line 48, the comment needs to change from 6 seconds to 12 as well. Just a minor change in the comment to keep it consistent with the code. :)

njordan77, 2018-08-08:
hi...
I'm quite familiar with ESPeasy on ESP12e items, also used Mega until February'18...now updated to recent releases but seem to run into same problem....

Sleep Enable tag is gone...now its an awake time...which i do not understand and its not mentioned in Wiki anywhere....whatever i set,....wake sleep works but no sensors are transmitted via MQTT..... any idea what is going on.....???? if no sleep (via awake 0) all works fine.....

TD-er, 2018-08-08:
I changed that setting to allow to set more than just a on/off toggle.
0 sec is indeed "always on" or "no sleep".
All other values represent the time awake, since a lot of people were having issues with some sensors needing more time to perform a sample.

So the set time is now something like "whatever happens, after N seconds go back to sleep"
You can force it to go back to sleep faster using rules.
The "sleep time" should make sense I guess. :)
The [wiki](https://www.letscontrolit.com/wiki/index.php/SleepMode) indeed needs an update on this.

There are indeed some issues regarding sleep and sensor readings and I am working on that.
Does your node have issues connecting to wifi after deep sleep?

njordan77, 2018-08-08:
no...i see the pings every 60secs for 12secs....so wifi should be fine....but no sensors at all are showing effects in MQTT-broker...and i use RSSI and ADC interally which should be easy +  a temp/hum Sensor si7021....

What i do not get...if you force to stay on a certain time.....like 12 secs you may be awake longer than really you need to...like what if the sensor takes only 2 secs.....up to before this update it seemed that it waits to send values and goes to sleep afterwards...which was perfect....now it seems another parameter to set which is hard to know and only experimentially to get right.....not sure if i miss some information not having the full picture....Cheers and Thanks Norbert

njordan77, 2018-08-08:
So i now set the awake time to 22sec and sleep time to 60secs (normally i use 600 - 10mins).....but ping works like it should for arround 20secs +/- i see the LWT connected and connection lost in mqtt broker...so MQTT sends something....but no sensor data.....

I even changed the sensor interval from 600 to 60....not sure if this has an impact....never knew exactly what this value means at all if i wake up from sleep.....is this interval value taken at all?....but still nothing works...no sensor data.

Can you tell me when did this "sleep enable" remove in the code...so i try a version before this change... Thanks

TD-er, 2018-08-08:
I am now -as we speak- looking into this and tomorrows build will have a fix for this.
My current state is that it sends all values as soon as MQTT has been established (when an MQTT controller is active).
Except the first plugin/task is not updating the value via MQTT. All other tasks work just fine and the same sensor (at pos 1) is sending fine to Domoticz HTTP.

So I will merge this now and go to bed. Tomorrow I will look into fixing that first MQTT value issue.

ghtester, 2018-08-09:
@njordan77 I don't use sleep features but if you wish to try some earlier version, I would recommend  mega-20180324 which is still based on core 2.3.0 and from my perspective is much stable & has less issues than later versions. The Sleep awake time parameter is already there but maybe it could work.

njordan77, 2018-08-09:
To let you know - went back to non-MEGA from MArch (the latest one) and did set AWAKE to 1sec.....
all works fine now - again. Also what was missing is now back - the SLEEP message in serial logging...that was not seen any more...

Subscribed to: /MyESP10/#
ADC  : Analog value: 1019 = 4.946
SYS  : -45.00
SLEEP: Entering deep sleep in 30 seconds.

TD-er, 2018-08-09:
That message is still present, I have seen it yesterday during tests.
The "30 second" message will only show when the last boot reason was anything else than a "wake from deep sleep". All other messages are something like "going to sleep now".

TD-er, 2019-10-10:
Is this still an issue?

TD-er, 2022-04-01:
Should be fixed by this PRs: 
- https://github.com/letscontrolit/ESPEasy/pull/3983

Please let me know if it is indeed fixed now... finally

closed 2022-04-01 by TD-er